hzeller/rpi-rgb-led-matrix #106: Slowdown Option for LED Image Viewer

viper2035, 2015-12-13:
Hi,
first of all thanks for the excelent work. I love the possibilities I have with the LED matrix. Is it maybe possible to at the  "DEFINES+=-DRGB_SLOWDOWN_GPIO=1" option to the LED Image Viewer? Because I've got some problems with flickering at the Image Viewer.

hzeller, 2015-12-13:
The image viewer just uses the library; so if the library is compiled with the slowdown option, the image viewer, that is linked against that library, uses that as well (you can try `make clean ; make led-image-viewer` to really make sure that this is happening, but the makefile rules already make sure that changes in the flags will require the led-image-viewer to be rebuilt).

How does the flickering you are observing with the image viewer but not with the demos look ? How many panels have you connected ?

viper2035, 2015-12-13:
Hi when I execute the make command it shows me that he is using the  "DEFINES+=-DRGB_SLOWDOWN_GPIO=1" option, but I still have a flickering that I dont have if I'm using the led-matrix command.
The flickering shows up pixels that should be black in a color of a pixels that is next to it. It's just one 32x16 panel.

hzeller, 2016-02-06:
The led-image-viewer is mostly a demo, feel free to whatever feature you'd like to see there (and maybe provide a pull-request if this sounds useful).

closed 2016-02-06 by hzeller