hzeller/rpi-rgb-led-matrix #107: Pipe ppm to led-matrix

radojko, 2015-12-15:
I'm trying to pipe ppm sequence to led-matrix, but it seems there is only possibility to load one image with scrolling( D option). What is the easiest way to pipe ppm sequence to the led-matrix?

hzeller, 2015-12-15:
You should probably just write a quick program that does it; not worth the hassle modifying the demo code.

radojko, 2015-12-15:
I'm not programmer, this is too complicated for me :(

I'm just trying to pipe avconv to get stream directly on rgb matrix. Also piping should be good and for connection with gstreamer ;)

Can you add pipe to the demo example? :)

hzeller, 2015-12-15:
On 14 December 2015 at 21:04, radojko <notifications@github.com> wrote:

> I'm not programmer, this is too complicated for me :(
>
> I'm just trying to pipe avconv to get stream directly on rgb matrix. Also
> piping should be good and for connection with gstreamer ;)
>
> Can you add pipe to the demo example? :)
>

Mmh, so how would it look like: you essentially concatenate PPMs and send
them through the pipe; and whenever one could be successfully decoded, it
is displayed ?

Might be an interesting option to add to the led-image-viewer. I consider
it :)

radojko, 2015-12-15:
PPM because I saw in demo, it can be and rawvideo or any avconv supported format. 

Yes, always displayed. 

I tested with  
avconv -i 'rtp://localhost:15001' -f image2pipe -vcodec ppm /tmp/test.ppm
and 
./led-matrix -D 1 /tmp/test.ppm -c 12 -P 3 -p 11

avconv in this case receive rtp/udp(ima my case  mjpeg)stream, convert them to ppm and pipe to the led matrix.

Any pipe option that works with avconv will pi awesome ;)

closed 2017-07-30 by hzeller

ccoenen, 2018-10-13:
I think what you're doing could now be done via `ledcat` found in `/examples-api-use`. I haven't tested it yet, though.