hzeller/rpi-rgb-led-matrix #108: dispite level shifters, display needs 3v3

siedi, 2015-12-17:
Hi,

I'm running a 32x64 panel connected to a rpi1 with a custom board using level shifters from 3v3 to 5v. The HUB16 do have a 74AHC245 "only", so according to your troubleshooting guide the shifters are needed. Now powering the panels with 5v just show random lines from top to bottom. Lowering the voltage to 3v5/3v3 the display works perfectly. I checked the output of the level shifters, they are working and output 4.96 something. I'm kind of stuck now, not sure why the panels work fine with 3v3 despite the shifter, but not with 5v.
Do you have any hint for me? My fears are when running with 3v3 it draws the power from the rpi through the signals.

Thanks 
Siedi

hzeller, 2015-12-18:
This is indeed strange. I can only think that the ground connection between your Pi/level shifter and the boards is not working properly. Do you have ground connected between these ?

Do you have a picture of your custom level shifter board ? Do you have 100nF buffer capacitors close to your level shifters ?

siedi, 2015-12-21:
Thanks for your advise. Finally I figured it out, I was kind of lazy, I had only 12 pins shifted, but needed 13, and this one (still 3v3) caused the problem.

closed 2015-12-21 by siedi