hzeller/rpi-rgb-led-matrix #109: Unused GPIO on 40 pin PI

Helbrax, 2015-12-28:
On the 40 pin Pi, you can support up the three displays.  If I only plan on running one, do I need to take any special consideration to use the remaining pins that would have been used for chains 2 and 3 as Inputs, such as buttons, etc?  Are the pins used by the remaining unused chains automatically set up when GPIO.Init() is called?

hzeller, 2015-12-29:
When you only use one chain, the other pins are not initialized for output, so your plan should work (I haven't tested it myself, but at least the pins kept free).

lanrat, 2015-12-29:
I also only use 1 display and have successfully been able to use the other unused pins.

hzeller, 2015-12-29:
Cool! Have fun hacking!
(are you building a game with 'game controller' buttons ?)

lanrat, 2015-12-29:
I use the extra pins to read the ambient light in the room with a photo-resistor to determine a good display brightness. My project is a "smart" clock that in addition to displaying the time and date; also displays the weather, ETA of nearby busses, current network bandwith, and more.
I was thinking if implementing a snake game that can be controlled over the network by a smartphone....
Its still a WIP but the source is available at https://github.com/lanrat/led-clock

Helbrax, 2015-12-29:
Thanks for the responses!  My end goal is a "voting" machine of sorts. I have the python code prototyped and my matrix should be here this week.  I just wanted to make sure it didn't default all pins to "on" when being run, and blowing up the pi or the matrix.

I assume that any python I write will need to use GPIO.BOARD instead of BCM so I can drive buttons and matrix at the same time within one python script?

closed 2016-01-01 by Helbrax