hzeller/rpi-rgb-led-matrix #112: Only 8 rows lighting up on 16x32 in 4 pairs

krishsgit, 2016-01-17:
I have an RPi 2 with 40 pins and have connected a 16x32 led matrix from adafruit to it with short wires directly from the Pi to the display using the instructions in the README.

Running led_matrix -r 16 lights up the LEDs. However only 4 pairs of rows light up. The other four pairs don't. Double checked the wiring and set SLOWDOWN for the GPIO to "1" and "2"

I'm not buffering the input yet. It is the direct 3.3V going into the LED matrix.
Could that be the reason that alternate pairs of rows don't light up?
Here is a picture of how things look right now.

[image]

maddogs, 2016-01-17:
I've got the exact same setup on my desk at the moment (direct connect from Pi to Matrix) and everything is fine.

Couple of things to check. When I first set one of these up last year, I did make the mistake of wiring up to the Output first. On my first few attempts of wiring up to the Input I reversed the connections on the Matrix, it wasn't until I saw the HUB75 notch on the left, I realised what I had the Matrix upside down.

davemaster, 2016-01-17:
Greetings,

Check the control pins, OE, CLK, LAT

Also, the order of your RGB GND pins, check it from the link  bellow

https://learn.sparkfun.com/tutorials/rgb-panel-hookup-guide

krishsgit, 2016-01-17:
Thanks @maddogs @davemaster 

The rewiring fixed it. 
Next stop is to try longer cables once my voltage shifter chips arrive.

closed 2016-01-17 by krishsgit