hzeller/rpi-rgb-led-matrix #114: Python binding graphics.Color brightness control

labajo, 2016-01-27:
Firstly thanks for this super project. Once I resolved the wiring problem, it was very simple to use the software and see the led panels light.

I managed to compile the python bindings in my raspberry pi 2 and I execute some python sample scripts.

I modified the color of the ```runtext.py``` script and I executed again the script and I noticed that I cannot handle the brightness of the colors.
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/python/samples/runtext.py#L15

For example, if I have the following statement:
```python
textColor = graphics.Color(255, 180, 100)
```
I can only see the red component in the led panels. Green and blue leds don't light.

The colors only light when the color value is equal to 255.
I tested the panels with ```led-matrix``` application and in that case I appreciate the brightness control. 
I don't know where I can start to see the problem because I have a very limited knowledge of C++.

maddogs, 2016-01-27:
So if you set the text colour to

`textColor = graphics.Color(0, 255, 0)`

You don't get Green text?

labajo, 2016-01-28:
In that case I have green color at 100% of brightness.
The color works when the color value has 255. You can work with combinations of colors if each color of the combination has 100% of brightness.
But for example if I use 
```python
textColor = graphics.Color(0, 180, 0)
```
or

```python
textColor = graphics.Color(0, 210, 0)
```
The led panel does not light.

I forgot to say that the -b argument in runtext.py execution does not work either. I think it is the same issue.

Perhaps It would be the operative system. I'm using Raspbian Jessie.

paulharfenmeister, 2016-01-28:
Hey, what is your command for starting the sample look like?

Try to set the ```-p``` option to 11.

maddogs, 2016-01-28:
I've just tried what I think you explained. So if you change a colour value from 255,  to say 210, the LEDs do not light at all?

I've just had a quick play with two rows of text, with colours codes as

```
		textColor = graphics.Color(0, 255, 0)
		textColor1 = graphics.Color(0, 30, 0)

```
And the brightness for each row is very different. Quick video of the result

https://youtu.be/_PPndQ3sXLE

This is with a Raspberry Pi 2 running Jessie & Passive Adaptor.

When I set the Green to 210 or 170, I really couldn't see much difference. I had to lower it below 100 before seeing a noticeable effect.

The -b flag worked fine on my machine to, dropping from -b 100 to -b50 to -b 10 

https://youtu.be/lhcpM7thC_I

labajo, 2016-01-29:
It was the -p argument. I though It handle the parallel chains, I see the led-matrix command is -P and not -p. I had -p to 1.  Thank you very much :smiley: . I close the issue.

closed 2016-01-29 by labajo