hzeller/rpi-rgb-led-matrix #115: PANEL_SWAP_G_B_

perl5577, 2016-01-28:
I have little probleme one pannel only are need PANEL_SWAP_G_B_

I  have need PANEL_SWAP_G_B_ only for chain 3

perl5577, 2016-01-28:
diff --git a/lib/framebuffer-internal.h b/lib/framebuffer-internal.h
index 7937b61..bc4d11d 100644
--- a/lib/framebuffer-internal.h
+++ b/lib/framebuffer-internal.h
@@ -175,8 +175,8 @@ private:
       // pins of type 'p0_r1' means 'first parallel chain, red-bit one'
       //                                 GPIO Header-pos
       unsigned int unused_0_1     : 2;  //  0..1  (only on RPi 1, Revision 1)
-      unsigned int p2_g1          : 1;  //  2 P1-03 (masks SDA when parallel=3)
-      unsigned int p2_b1          : 1;  //  3 P1-05 (masks SCL when parallel=3)
+      unsigned int p2_b1          : 1;  //  2 P1-03 (masks SDA when parallel=3)    < Inverse
+      unsigned int p2_g1          : 1;  //  3 P1-05 (masks SCL when parallel=3)    < Inverse
       unsigned int strobe         : 1;  //  4 P1-07
       unsigned int p1_g1          : 1;  //  5 P1-29 (only on A+/B+/Pi2)
       unsigned int p1_b1          : 1;  //  6 P1-31 (only on A+/B+/Pi2)
@@ -191,14 +191,14 @@ private:
       unsigned int p1_g2          : 1;  // 13 P1-33 (only on A+/B+/Pi2)
       unsigned int p2_r1          : 1;  // 14 P1-08 (masks TxD when parallel=3)
       unsigned int unused_15      : 1;  // 15 P1-10 (RxD) - kept free.
-      unsigned int p2_g2          : 1;  // 16 P1-36 (only on A+/B+/Pi2)
+      unsigned int p2_b2          : 1;  // 16 P1-36 (only on A+/B+/Pi2)      < Inverse
 
       unsigned int clock          : 1;  // 17 P1-11
 
       unsigned int output_enable  : 1;  // 18 P1-12 (PWM pin: our timing)
       unsigned int p1_r2          : 1;  // 19 P1-35 (only on A+/B+/Pi2)
       unsigned int p1_b2          : 1;  // 20 P1-38 (only on A+/B+/Pi2)
-      unsigned int p2_b2          : 1;  // 21 P1-40 (only on A+/B+/Pi2)
+      unsigned int p2_g2          : 1;  // 21 P1-40 (only on A+/B+/Pi2)    < Inverse
 
       unsigned int a              : 1;  // 22 P1-15  // row bits.
       unsigned int b              : 1;  // 23 P1-16

If make fixing with at :)

hzeller, 2016-03-08:
That will certainly do it :)

closed 2016-03-08 by hzeller