hzeller/rpi-rgb-led-matrix #116: Raspberry pi 2

cbogithub, 2016-02-09:
I have 1 led board rbg with i have just buy at :http://www.aliexpress.com/store/product/Leeman-P5-indoor-rgb-LED-module-High-Precision-p3-p4-p5-HD-SMD-Full-Color/1239156_32322111301.html   
and i tried tut https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/wiring-the-display   with power 5v - 40A

 GND on display to GND on the Pi (blue or black)
R1 on display to GPIO 17 on the Pi (red)
G1 on display to GPIO 18 on the Pi (green)
B1 on display to GPIO 22 on the Pi (blue)
R2 on display to GPIO 23 on the Pi (red)
G2 on display to GPIO 24 on the Pi (green)
B2 on display to GPIO 25 on the Pi (blue)
A on display to GPIO 7 on the Pi (yellow or white)
B on display to GPIO 8 on the Pi (yellow or white)
C on display to GPIO 9 on the Pi (yellow or white)
OE on display to GPIO 2 on the Pi (brown)
CLK on display to GPIO 3 on the Pi (orange)
LAT on display to GPIO 4 on the Pi (yellow)

but not working ??? no anything show on led matrix board (i used command :`./led-matrix -t 10 -D 1 runtext.ppm`)

hzeller, 2016-02-09:
Please read the Readme of this project, NOT the outdated Adafruit article
which doesn't reflect the current wiring.

cbogithub, 2016-02-09:
OK, and now it working but  it only show 2 block when run : ` ./led-matrix -t 10 -D 1 runtext.ppm`
http://i.imgur.com/HMrtnov.jp

maddogs, 2016-02-09:
try updating your command 

sudo ./led-matrix -t 10 -D 1 runtext.ppm -r 32 -c 2

cbogithub, 2016-02-09:
still not working :http://i.imgur.com/K1nxrme.jpg
it show 4 block : red,red, white, green ....OMG ! i tried check switches --> ok,

closed 2017-07-30 by hzeller