hzeller/rpi-rgb-led-matrix #117: build led-image-viewer error

cbogithub, 2016-02-09:
`root@raspberrypi:~/Led/rpi-rgb-led-matrix# apt-get install webp
Reading package lists... Done
Building dependency tree
Reading state information... Done
webp is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@raspberrypi:~/Led/rpi-rgb-led-matrix# make led-image-viewer
make -C lib
make[1]: Entering directory '/root/Led/rpi-rgb-led-matrix/lib'
make[1]: Leaving directory '/root/Led/rpi-rgb-led-matrix/lib'
g++ -Wall -O3 -g led-image-viewer.o -o led-image-viewer -Llib -lrgbmatrix -lrt -lm -lpthread `GraphicsMagick++-config --ldflags --libs`
/usr/bin/ld: cannot find -lwebp
collect2: error: ld returned 1 exit status
Makefile:35: recipe for target 'led-image-viewer' failed
make: *** [led-image-viewer] Error 1
root@raspberrypi:~/Led/rpi-rgb-led-matrix#

hzeller, 2016-02-09:
You want to compile something, so webp-dev is probably what you want to
install.

timothystewart6, 2016-02-19:
I am having the same issue.  I can't seem to find webp-dev in aptitude

```
➜  rpi-rgb-led-matrix git:(master) ✗ sudo apt-get install webp-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package webp-dev
```

timothystewart6, 2016-02-19:
found it.

```
➜  rpi-rgb-led-matrix git:(master) ✗ sudo apt-get install libwebp-dev
```

commit e1bc8b2c (2016-03-08): o The graphicsmagick maintainers forgot a dependency, so add it
  in the doc to add. Fixes #117
closed 2016-03-08 by hzeller