hzeller/rpi-rgb-led-matrix #118: G1 not working on Raspberry Pi 1 Rev 1

tazfriend, 2016-02-22:
On a Rapsberry Pi A/B PCB Rev 1 (September 2012 or earlier), pin 13 of the GPIO connector is connected to GPIO 21 not GPIO 27. Could one change this with a compile flag.

hzeller, 2016-02-22:
If you define RGB_CLASSIC_PINOUT and ONLY_SINGLE_CHAIN then output-enable and clock will be on P1-03 and P1-05

See this:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer-internal.h#L121

.. for wiring.

davemaster, 2016-02-22:
Greetings,

Try replacing RGB cables.

hzeller, 2016-03-08:
Did it work @tazfriend ?

crazyduck, 2016-05-30:
Dirty workaround for RPi1rev1, as PIN13 is GPIO27(RPi1) replace in 
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/framebuffer-internal.h#L210

unsigned int p2_b2          : 1;  // 21 P1-40 (only on A+/B+/Pi2)

with

unsigned int p0_g1          : 1;  // 21 P1-40 (only on A+/B+/Pi2)

hzeller, 2016-06-14:
No response by @tazfriend - but anyway, whoever stumbles over this, the suggstion above (setting RGB_CLASSIC_PINOUT and ONLY_SINGLE_CHAIN) should work. Closing now.

closed 2016-06-14 by hzeller