hzeller/rpi-rgb-led-matrix #119: Vertical image scroller

daker, 2016-02-27:
Hello,

I am trying to implement a vertical image scroller by modifying the horizantal scroller, can you please tell what's i am missing ?

Thanks!

```c++
class ImageScroller2 : public ThreadedCanvasManipulator {
public:
  ImageScroller2(RGBMatrix *m, int scroll_jumps, int scroll_ms = 30)
    : ThreadedCanvasManipulator(m), scroll_jumps_(scroll_jumps),
      scroll_ms_(scroll_ms),
      vertical_position_(0),
      matrix_(m) {
      offscreen_ = matrix_->CreateFrameCanvas();
  }

  virtual ~ImageScroller2() {
    Stop();
    WaitStopped();   // only now it is safe to delete our instance variables.
  }

  bool LoadPPM(const char *filename) {
    FILE *f = fopen(filename, "r");
    // check if file exists
    if (f == NULL && access(filename, F_OK) == -1) {
      fprintf(stderr, "File \"%s\" doesn't exist\n", filename);
      return false;
    }
    if (f == NULL) return false;
    char header_buf[256];
    const char *line = ReadLine(f, header_buf, sizeof(header_buf));
#define EXIT_WITH_MSG(m) { fprintf(stderr, "%s: %s |%s", filename, m, line); \
      fclose(f); return false; }
    if (sscanf(line, "P6 ") == EOF)
      EXIT_WITH_MSG("Can only handle P6 as PPM type.");
    line = ReadLine(f, header_buf, sizeof(header_buf));
    int new_width, new_height;
    if (!line || sscanf(line, "%d %d ", &new_width, &new_height) != 2)
      EXIT_WITH_MSG("Width/height expected");
    int value;
    line = ReadLine(f, header_buf, sizeof(header_buf));
    if (!line || sscanf(line, "%d ", &value) != 1 || value != 255)
      EXIT_WITH_MSG("Only 255 for maxval allowed.");
    const size_t pixel_count = new_width * new_height;
    Pixel *new_image = new Pixel [ pixel_count ];
    assert(sizeof(Pixel) == 3);   // we make that assumption.
    if (fread(new_image, sizeof(Pixel), pixel_count, f) != pixel_count) {
      line = "";
      EXIT_WITH_MSG("Not enough pixels read.");
    }
#undef EXIT_WITH_MSG
    fclose(f);
    fprintf(stderr, "Read image '%s' with %dx%d\n", filename,
            new_width, new_height);
    vertical_position_ = 0;
    MutexLock l(&mutex_new_image_);
    new_image_.Delete();  // in case we reload faster than is picked up
    new_image_.image = new_image;
    new_image_.width = new_width;
    new_image_.height = new_height;
    return true;
  }

  void Run() {
    const int screen_height = matrix_->transformer()->Transform(offscreen_)->height();
    const int screen_width = matrix_->transformer()->Transform(offscreen_)->width();
    while (running()) {
      {
        MutexLock l(&mutex_new_image_);
        if (new_image_.IsValid()) {
          current_image_.Delete();
          current_image_ = new_image_;
          new_image_.Reset();
        }
      }
      if (!current_image_.IsValid()) {
        usleep(100 * 1000);
        continue;
      }
      for (int x = 0; x < screen_width; ++x) {
        for (int y = 0; y < screen_height; ++y) {
          const Pixel &p = current_image_.getPixel(x, (vertical_position_ + y) % current_image_.height);
          matrix_->transformer()->Transform(offscreen_)->SetPixel(x, y, p.red, p.green, p.blue);
        }
      }
      offscreen_ = matrix_->SwapOnVSync(offscreen_);
      vertical_position_ += scroll_jumps_;
      if (vertical_position_ < 0) vertical_position_ = current_image_.height;
      if (scroll_ms_ <= 0) {
        // No scrolling. We don't need the image anymore.
        current_image_.Delete();
      } else {
        usleep(scroll_ms_ * 1000);
      }
    }
  }

private:
  struct Pixel {
    Pixel() : red(0), green(0), blue(0){}
    uint8_t red;
    uint8_t green;
    uint8_t blue;
  };

  struct Image {
    Image() : width(-1), height(-1), image(NULL) {}
    ~Image() { Delete(); }
    void Delete() { delete [] image; Reset(); }
    void Reset() { image = NULL; width = -1; height = -1; }
    inline bool IsValid() { return image && height > 0 && width > 0; }
    const Pixel &getPixel(int x, int y) {
      static Pixel black;
      if (x < 0 || x >= width || y < 0 || y >= height) return black;
      return image[x + height * y];
    }

    int width;
    int height;
    Pixel *image;
  };

  // Read line, skip comments.
  char *ReadLine(FILE *f, char *buffer, size_t len) {
    char *result;
    do {
      result = fgets(buffer, len, f);
    } while (result != NULL && result[0] == '#');
    return result;
  }

  const int scroll_jumps_;
  const int scroll_ms_;

  // Current image is only manipulated in our thread.
  Image current_image_;

  // New image can be loaded from another thread, then taken over in main thread.
  Mutex mutex_new_image_;
  Image new_image_;

  int32_t vertical_position_;

  RGBMatrix* matrix_;
  FrameCanvas* offscreen_;
};
```

hzeller, 2016-06-14:
I have not run this code but it roughly looks like what you should do. What is not working ? Did you figure it out in the meantime ?

Guillaumecht, 2016-06-24:
Hi, 

I would like to do the same. 
i have 4 pannels of 32*32 put together (to build a line), and i want to display text and scroll it vertically. 
Did you succeed to build an vertical scroller ? 

Guillaume

hzeller, 2016-06-24:
For that, you probably want to directly use the text feature instead of preparing an image to be scrolled. Makes it much simpler when you want to change the text :)

See the `text-example.cc` for a code example how to use text.

Guillaumecht, 2016-06-24:
I checked `runtext.py`, and it seems very easy to do what i want with Python. I more familiar with Python than C++. Thank you for your fast answer.

hzeller, 2016-06-24:
Cool!

Guillaumecht, 2016-06-25:
I'am a bit lost; i understood runtest.py, but not able to find any documentation to achieve what i want.
Where can in find the documentation ? 

i try to display text on display 128x32, so text should scroll vertically and need to be cut on multi-line. 
Can you help me ?

hzeller, 2016-06-25:
The Python API just wraps the C++ API; the documentation is there
 https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L73

You have to cut your text yourself of course on the multi-line boundaries. Then write each line under each other with a y-offset of your choosing (typically you want something like at least font.height). Do that in a offscreen swap-buffer and change the starting y position in every loop.
For simplicity, just write the full text every time, it will be clipped at your panel borders.

Guillaumecht, 2016-06-25:
Thank you very much for your hepl.
One last more think, I tried to change the textsize and, if possible make text in bold (or a way to display letter biggest). Is it possible ?

hzeller, 2016-06-25:
There are a few fonts in the fonts/ directory, some have 'B' at the end of the name, such as 7x14B.bdf - these are the bold fonts.

closed 2017-07-30 by hzeller