hzeller/rpi-rgb-led-matrix #121: Python3: No module named 'core'

rasusmilch, 2016-03-13:
I am attempting to use this with a 64x32 panel connected to a Raspberry Pi 2 B.

When running a `make build-python` I see it is attempting to build and install the Python2.7 code.
If I enter the python directory and manually run a `python3 setup.py build/install` it builds and installs the python3 modules. However, using the simple example of drawing a red rectangle
```python
from rgbmatrix import RGBMatrix
import time

rows = 32
chains = 2
parallel = 1
myMatrix = RGBMatrix(rows, chains, parallel)
myMatrix.Fill(255, 0, 0)
time.sleep(5)
myMatrix.Clear()
```
gives me
```
Traceback (most recent call last):
  File "led_Matrix.py", line 8, in <module>
    from rgbmatrix import RGBMatrix
  File "/usr/local/lib/python3.4/dist-packages/rgbmatrix/__init__.py", line 4, in <module>
    from core import RGBMatrix, FrameCanvas
ImportError: No module named 'core'
```

I have attempted to find where in the makefile it selects python2.7 to compile, but was unsuccessful (I haven't learned makefiles yet.) I thought about removing python2.7, but then it removes a lot of other things I need (and these python modules need, such as python3-cxx-dev I believe.)
```
The following packages were automatically installed and are no longer required:
  libpython-dev libpython2.7-dev screen tmux
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  byobu python python-cxx-dev python-dev python-newt python-ownet python-rpi.gpio python-smbus
  python2.7 python2.7-dev python3-cxx-dev
```
I then thought it was a naming error since it calls the compiled modules `core.cpython-34m.so` and `graphics.cpython-34m.so` but naming them to `core.so` and `graphics.so` like the python2.7 modules did not work.

If I run the code as python2.7, everything seems to work (at least with the simple example code.) However, I am writing a Home Automation system with extensive python3.4 code already written.

In all, this seems like a simple thing I should be able to figure out, but combining Python, C++ and make has me befuddled.

hzeller, 2016-03-13:
Adding @Saij as he knows the Python stuff.

Saij, 2016-03-14:
I didn't test the code with Python 3.
But I will test it as soon as my testsystem is working again (but might take a while)

rasusmilch, 2016-03-14:
A litle closer. Seems to be a path issue. If I copy the python file `led_Matrix.py` to the module location `/usr/local/lib/python3.4/dist-packages/rgbmatrix` and run it, it correctly finds the `core.cpython-34m.so` and `graphics.cpython-34m.so`.

However, running
```
# python3 -c 'import sys; print(sys.path)'       
['', '/usr/lib/python3.4', '/usr/lib/python3.4/plat-arm-linux-gnueabihf', '/usr/lib/python3.4/lib-dynload', '/usr/local/lib/python3.4/dist-packages', '/usr/lib/python3/dist-packages']
```
Shows me that the path should be included to find the module. The rgbmatrix directory does have an `__init__.py` file, so it should find it (as my googling suggested without this file Python won't traverse the directory.)

Am I missing something obvious?

Saij, 2016-03-14:
Like I said I will take a look into it. Just need some time to get my Pi up and running again and then just have to get fluent in Python 3 ^.^

rasusmilch, 2016-03-14:
Thanks. I'm not attempting to hound anyone. I appreciate the work that went into this, both yourself and hzeller. 

I'm just doing some investigating and thought I would offer what I figure out to reduce your workload.

Saij, 2016-03-14:
If you like you can investigate and fix the problem ^.^ I would appreciate that ^.^

rasusmilch, 2016-03-14:
You guys are way ahead of me. I admit to going back to school for Computer Science, but it would take me a long time to get where everyone else is at. I'm not saying someday I won't look at it, but my skill level is not quite there yet.
Also need to buckle down for the rest of my classes. Depending on the situation I may have time this summer. However, I could never have done this melding of Python and C++ already done.
There is no hurry, I placed the libraries within the same directory as my Python3 file and it now finds them. Note that in methods calling things with text strings you need to use binascii.a2b_qp(TEXT STRING) to convert strings to bytes, as Python3 now uses Unicode strings instead of Python2.x byte strings.

Saij, 2016-03-15:
Thanks for the information with the string handling. Might come in handy ^.^

rasusmilch, 2016-03-16:
For your reference, here is where my Ah-ha came from regarding TypeErrors with fonts and text.
[Strings in Python](https://wiki.python.org/moin/BytesStr)

Saij, 2016-05-24:
Quick Fix:
in /usr/local/lib/python3.4/dist-packages/rgbmatrix/__init__.py just put a dot (.) before the word "core"

That way it'll work.
I'll fix the file the next days and also try to update the Makefile to choose between python2 and python3

closed 2016-06-14 by hzeller