hzeller/rpi-rgb-led-matrix #123: Can we chained many row

lucas-pouchot, 2016-03-22:
HI, I've linked 19 rows of 4 16x32 panels on a Raspberry Pi 2 with the classic pinout like this : 

__ Rasperry
|
|_ P01 _ P02 _ P03 _ P04_
______________________|
|
|_ P05 _ P06 _ P07 _ P08 _
______________________|
|
|_ P09 _ P10 _ P11 _ P12 _

etc up to P76

It is possible to use your library to draw a ppm with this resolution (304*128) with led-matrix ?
If not, can you tell me how to do ?

This it's what I have when I do : 
`./led-matrix -r 16 -P 1 -c 76 -D 1 /tmp/sample.ppm`

[image]

Thanks in advance for your help.

hzeller, 2016-03-24:
Ha that is neat! Do you have a picture of the back ?

It should be possible in theory, but in practice, there might be some hurdles.

You really want to use three parallel chains though to have higher refresh-rate and shorter chains. Single chain is really only needed if you have an old Raspberry Pi 1 that doesn't have enough outputs.

Here are some things to keep in mind:

 * You need to start with a 3.3V  -> 5V level converter to have a nice, clean signal to start with (you didn't mention if you had one).
 * There are long lines between the rows (like P4 and P5). This could lead to problems with signal integrity. In general, keep data lines as short as possible.
 * Timing issues are likely to crop up between panels, depending on how they pass on the clock signal for instance. You might have to slow down the data (see RGB_SLOWDOWN_GPIO). You might need even more than is possible with `-DRGB_SLOWDOWN_GPIO=2`, so have a look at include/gpio.h and do that manually (there are some commands that need to be repeated). If there are issues, you need to understand how the basic data should look like and measure with an oscilloscope. It might be a good idea to check how far data and clock are out-of-sync after each row and possibly add gates to selectively delay lines.
 * To map the coordinates properly, you need to do some mapping using the transformer feature ( include/transformer.h )
 * the refresh rate with such a long chain will not be so good. With less pwm-bits, it can be improved at the expense of color-richness. Use `-p 1` for instance for just one bit, i.e. 8 colors.
 * Also: you really don't want to use the classic pinout.

I have a 160x96 pixel display running with three parallel chains of 5 panels with 32x32 pixels (see picture from the back in README). Which is a little less than half the pixels you want to show, and it works fine on a Raspberry Pi 2 with ~ 200Hz refresh rate. Here, only 5 panels are per chain but it should be possible to extend the chains quite a bit. However the main problem might be timing problems that add up between panels.

So what you want to do is

 * Start with building an active adapter like this one https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3
 * Chain 26 of these panels on each of the chains.
 * If possible, go back-and-forth rather than starting from the start with each chain. That way, you keep cables short. So

```
    ->  -> -> -> |
    <- <- <- <-  v
```
  Well, sometimes that is not so easy as the panels look different when upside down.
 * Test if things work. You might need to check with the oscilloscope after each row how the signal
   integrity is (it really depends on the quality of the LED panels).
 * You still might need to slow down the data rate (see gpio.h mention above).
 * Make sure to provide proper power. I estimate that you need in the order of 120A at 5V to power this. I'd probably get three 40A-60A power supplies. Make sure to connect ground between these with beefy connections and keep power cables short (also prefer star-connection to bus-connection). Don't connect the 5V lines with each other (tiny voltage differences would otherwise create huge equalization currents), but connect each of them to a subset of displays (not sure, you might already doing that, hard to tell from the picture).
 * Make sure that the combined ground-connection is also going to the Pi.
 * Put a Low-ESR cpacitor (something in the order of 2200uF or more) directly on the power input of _each_ panel.
 * Measure the voltage on the panel; You want to not have too much voltage drop from the source of the power supply. If you have <= 250mV voltage drop from supply to panel, then it might help to increase the voltage on the supply to 5.25V (but not much more as otherwise you have an over-voltage when the display is dark). Calculate the needed thickness of the wires, you need quite some beefy ones.
 * Measure the power supply with a scope in AC mode and see how noisy it is.
 * (In general, read the 'A word about power' section of the README).
 * If you want to use this to display a large picture (video or image), write a Transformer class that does a proper coordinate mapping for your physical layout, so that the pixels are presented as 304x128 'display'.

maddogs, 2016-03-24:
I'm interested in seeing this working! If you dont have an Active level shifter, I can post you a spare Active board, 40 pin header and three 16 pin sockets. That I have left over.

You'll need to get the four 474HCT245 SOICs, caps and resistors and solder it together, but they're pretty cheap readily available.

Saij, 2016-03-24:
I still wonder where you buy those panels for cheap... Want to build something similar - but with a price of around 60$ per panel it is too much (especially with shipping to Germany).

As for the Active Level Shifer: do you think it would be a good idea to place the levelshifter direclty onto the panels? You might need more level shifters but you could compensate voltage drop on the datalines in my opinion.

maddogs, 2016-03-24:
We got 10 active boards made in Chine here

http://www.pcbway.com/

For $38 inc DHL Shipping to the UK

Saij, 2016-03-24:
I didn't mean the active boards ^.^ They are cheap ^.^
But the LED panels are expensive...

maddogs, 2016-03-24:
Ah! Miss read your post :) You can get the panels a lot cheaper if you don't mind a bit of a wait for delivery. a 64x32 P4 you can get for $16.82 shipped a P5 $21.58 from aliexpress.com http://goo.gl/IUwzdO

We got 10 64x32 P5s for just $130 after contacting the seller and asking for a discount. They took 4 weeks to arrive  to the UK though.

And thanks for all the work you've put into the Python code, it's very much appreciated :)

Saij, 2016-03-24:
Thank you for the link.

hzeller, 2016-03-24:
@Saij you only need a level shifter once at the 'entrance'. Once in a panel and passing on to the next panel, they are already buffered by the panels. It is actually best to have the level shifter close to the Pi.

lucas-pouchot, 2016-03-29:
Hi, 
The first data cable was deficient, I changed it and it works well now, with 76 panels in one row, the picture is flickering, but with twenty panels, it is OK.
So I'm waiting receive the active adapter, I will write a transformer class and post new pictures when it will work fine.

maddogs, 2016-03-29:
@lucas-pouchot Did you order some, or did you want me to post the one I have?

Saad-Malik, 2016-04-26:
Where can i get the active panel from ?

hzeller, 2016-04-26:
The gerber files are posted (also a link to OshPark), but you have to solder it up yourself.

Saad-Malik, 2016-04-26:
Alright thanks.. I want to play a video on the led modules. I have 16*32 tiles. Can there be a python script that can show the hdmi output on the led modules. Please help me out. I need to submit a project in 10 days.

Saad-Malik, 2016-04-26:
I have managed to store the videos into the raspberry pi folder through a python script. The scripts downloads the videos from my server and store them in the raspberry pi. Videos are in mp4 format.  Now i want to run these videos on the led modules ?

Saad-Malik, 2016-04-26:
i tried this tutorial but it shows me red and green lines

 https://learn.adafruit.com/raspberry-pi-led-matrix-display

hzeller, 2016-04-26:
You can very simply utilize the ffmpeg library to play videos. It is probably 1h of work left as an exercise for the reader :)

Saad-Malik, 2016-04-26:
Can you please tell me the steps :D

closed 2016-05-06 by hzeller

lucas-pouchot, 2016-06-28:
HI, I finally wrote my transformer class, using the new wiring schema : 3 parallel rows of 36 panels

[image]

[image]

Obviously, the CPU is overloaded and the picture is flickering. But the result is impressive :)

hzeller, 2016-06-29:
Wow, that is spectacular and very impressive and beautiful!

You might get a little faster update rates with the the PWM set to less bits, but oviously that means less colors. You could try to operate it a little faster by not having `-DRGB_SLOWDOWN_GPIO=1` in lib/Makefile, but many displays can not deal with the speed.

You could also try tweaking `kBaseTimeNanos` in `lib/framebuffer.cc` a little.

(I have some idea to actually provide six or maybe more parallel chains by utilizing that excess speed, but that is not done yet).

More boring ways of course would be to use two Raspberry Pi's and have them synchronize a framebuffer via UDP packets.

Can I get permission to use your pictures in the 'fun' section at the bottom of the README ?