hzeller/rpi-rgb-led-matrix #124: Interference/flickering in animation

genregasm, 2016-03-24:
Not quite sure what's going on here. I'm using the new pin layout, because (as accurately described somewhere on this code) the adafruit article is outdated garbage. Using a Pi 2 B+. Issue occurs regardless of whether the DRGB_SLOWDOW_GPIO is active and what it's set at is irrelevant. Do I need a shifter chip? The cables are very short (as shown in video) and I checked them twice.

Video:
https://www.youtube.com/watch?v=70QwhFAC7yE

KentWalker, 2016-03-24:
I would consider level shifting since the Raspberry Pi GPIO is outputting 3.3 volt levels whereas your panel will be expecting 5 volt levels.  Henner has kindly put up a PCB design up on OSH Park which makes things easy although you can wire everything on veroboard.  

Use 74HCT245 or similar buffer ics (you must use 74HCTxxx, don't use 74HCxxx ics, they are a different family and won't work!).  Also fit 100nF capacitors across the supply pins to each ic.

The veroboard method does work and although there are a gazillion wire jumpers I had no problem with ghosting.  Just make sure to clean your board (whichever assembly method you use) to remove all solder flux.

Helbrax, 2016-03-25:
KentWalker is correct.  You need to level shift.  I bought 5 74HCT245s from digi-key for quite cheap.  Without them, I was getting similar results:
https://www.youtube.com/watch?v=fnykltA_qps

With them, there is no flicker.  Wiring them up is super simple as well.

genregasm, 2016-03-29:
Thank you gentlemen! What is the best way to connect them without hefty soldering or a breadboard? Which one is pin 1?

Helbrax, 2016-03-29:
The chip should have a dip/divot indicating the "top".  This is how I currently have mine set up.  Please note that this meets my needs, and may not meet yours depending on what you are trying to accomplish, but this will PROBABLY work for you.

http://i.imgur.com/ACVgfWc.gif

genregasm, 2016-04-01:
I apologize if my questions seem so naive. Can I buy the adapter board somewhere as opposed to...printing and building it myself? I got the chips, and they are _TINY_. I was under the impression I put these on the pin wires between the RPi and LED panels, but they're so small.

http://imgur.com/1pzaBe1
https://www.digikey.com/product-detail/en/nxp-semiconductors/74HCT245PW,112/568-2809-5-ND/1023084

genregasm, 2016-04-01:
That's what I get for not reading everything before continuing. Rule no.1.

hzeller, 2016-04-01:
Don't get the TSSOP chip, the SOIC is the one you need. That is solderable easily.
If you do it on a breadboard, then get the DIP package.

I am not selling ready-build breakout boards, but if there is enough demand, I might.

hzeller, 2016-04-01:
The board itself is published on OshPark https://oshpark.com/shared_projects/QIA9YHZH
... but you have to solder it yourself of course.

genregasm, 2016-04-18:
After ordering the wrong parts and re-evaluating my plan, I have a couple more questions.
-Does it matter what amp 74HCT245 chips? There is 4mA, 6mA, and 7.2mA
http://www.digikey.com/product-search/en/integrated-circuits-ics/logic-buffers-drivers-receivers-transceivers/2556308?k=74HCT245&pv69=3&pv16=6526&pv1291=200&mnonly=0&newproducts=0&ColumnSort=0&page=1&quantity=0&ptm=0&fid=0&pageSize=25
-Can I get resistors and capacitors from Digi-Key? I was having trouble finding the right ones.

hzeller, 2016-04-18:
* As long as they are HCT245 this is fine. The current is not significant in this case.
 * Search for [ceramic chip capacitor]. The value you are looking for is 100nF or sometimes written as 0.1uF (these are needed one for each HCT chip).
 * Same for resistors; they are called chip resistor (package 0805). The resistor there is somewhat optional (not needed for functionality, just helps to have not random LEDs go on when the software is not active)

closed 2016-05-06 by hzeller