hzeller/rpi-rgb-led-matrix #125: python.h not found when doing 'make build-python' for using python samples

delectatronics, 2016-03-30:
First of all please accept my thanks for a great contribution - it definitely helps out newbies like myself to this type of venture.
I have a RPi3 B+ using two chained 64x32 RGB LED panels from Adafruit to make a 64x64 display.  I have successfully used and installed the rpi-rgb-led-matrix and various of the examples without any issue.
However when trying to do the python builds I am getting the following error message
> rgbmatrix/core.cpp:4:20: fatal error: python.h: no such file or directory
  #include python.h
  compilation terminated
<

I believe this is trying to compile under Python 2.7
If anyone can point me in the right direction as to whether there is something else I have not yet installed or have perhaps done wrong, that would be greatly appreciated.

Please advise if there is additional diagnostic information required 

Many thanks

Saij, 2016-03-30:
You are right. Currently it is only working for Python 2.7 (but I on it to get it up and running on Python 3 ^^)

As for your problem, do you have installed python2.7-dev?
If not, try running following command:

`sudo apt-get install python2.7-dev`

also please be sure that `python` resolves to the corrent python version.
If `python` resolves to python3 just try following command:
`sudo update-alternatives --install /usr/bin/python python /usr/bin/python2.7 1`

delectatronics, 2016-03-30:
Thank you Saij - that worked perfectly.

I just did the **sudo apt-get install python2.7-dev** and then did the builds again and all of the displays worked.
Now comes the fun part!

closed 2016-03-30 by delectatronics