hzeller/rpi-rgb-led-matrix #127: 2 more parallel chains with Compute Module

currentlysober, 2016-04-12:
Dear Henner,

according to our conversation, I was trying to run 2 more chains (4th and 5th) using Compute Module. The new CMv3 is going to be released soon so once we get prepared, we will be able to show nice images on 5 chains.
I decided to use as following:
gpio30 -> R1
gpio30 -> R1
gpio30 -> R1
gpio30 -> R1
gpio30 -> R1
gpio30 -> R1
gpio30 -> R1
gpio30 -> R1
gpio30 -> R1
gpio30 -> R1

hzeller, 2016-05-05:
You are going beyond the 32 bits, so you need to change the gpio operations to do 64 bits.

currentlysober, 2016-05-08:
Hi!
The problem is that your library is too complicated for me :P Of course I tried, I have put changes and here I uploaded:
https://github.com/currentlysober/led_temp_files
but still there is no result. I am sure I missed many places but when I changed more uint32-s into 64, I couldn't see anything on the screen (even in chains 1-3). 
Moreover, I had an error:
static void rgb_matrix::internal::Framebuffer::InitGPIO(rgb_matrix::GPIO*, int): Assertion `result == b.raw' failed.
So I changed into
assert(result == (uint32_t)b.raw);
Could you please take a look and give some advice? I'll be grateful. Then we can merge it into official library. Thanks!

closed 2016-06-27 by currentlysober