hzeller/rpi-rgb-led-matrix #128: Working with Python

bagumondigi, 2016-04-13:
Hello, i'm very new in OOP using python.
in line 'from samplebase import SampleBase' i know it will import class SampleBase in samplebase.py. I can't find rgbmatrix.py in line 'from rgbmatrix import RGBMatrix' but it still work, why ?
and what's a link between python and these .cpp files in rgbmatrix folder ? and how it was generated ?

Saij, 2016-04-13:
There will be no rgbmatrix.py because it is a compiled module (and not a python file). It is created from the .cpp files and these files are created with Cython as a connection between Python and C++

bagumondigi, 2016-04-13:
Thank you for quick reply.
So, what file define the RGBMatrix class ? I want to understand the code, Should i read the 'core.cpp', 'graphic.cpp' or something else ?

Saij, 2016-04-13:
If you want to understand how the compiled module is made, you should read something about Cython.

The module is just C++ Code which is compiled in a way that Python understands. That's why there is no rgbmatrix.py but Python knows it.

bagumondigi, 2016-05-07:
@Saij Is it possible to use class LargeSquare64x64Transformer or any Transformer ? thank you

bagumondigi, 2016-05-10:
Hello,
Is it possible to use class LargeSquare64x64Transformer or any Transformer in Python ? thank you

Saij, 2016-05-24:
I will try to implement transformers in python soon.
For more please take a look at issue #101

closed 2016-05-24 by bagumondigi