hzeller/rpi-rgb-led-matrix #130: Schema for wiring like the passive adapter

lucas-pouchot, 2016-04-19:
Hi, I would like test the wiring of the passive adapter but I don't found the schema like for the active adapter here : 
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/img/active3-schematic.png

Can I just to connect them like the active adapter, without the four controllers ?
If no, could you share a schema of the wiring for the passive board ?

Thanks in advance.

lucas-pouchot, 2016-04-22:
Hi, I've follow this table to wiring 3 row in parallel with 26 chained panels: 
[image]

And I run `./led-matrix -D 1 -r 16 -c 26 -P 3 test.ppm`
but only the first row display something
Someone would have any idea to why?

hzeller, 2016-04-22:
If the test.ppm is large enough, then I assume that there is some wiring problem with the other panels. There are some wires (OE, strobe, clock. A, B..) that have to be wired to all panels. Also make sure that ground is properly connected (and yes, the schematic for the passive converter is exactly the same as this table).

Then there is the problem with 3.3V/5V level. If your panel needs 5V logic level, then 3.3V will not do it and you'll need to have a level converter.

Also make sure to connect all the grounds of the panels together and with the ground of the Pi.

hzeller, 2016-05-05:
Did this end up working ?

closed 2016-05-06 by hzeller