hzeller/rpi-rgb-led-matrix #132: Led matrix buying guide

soyxan, 2016-05-03:
First of all sorry as I think this post is not an Issue but I cannot find a more general forum to ask for help.

I am involved in a project and I am looking for a proper led matrix. I have found the following one in Aliexpress:

http://www.aliexpress.com/item/high-resolution-led-matrix-display-module-p5-smd-indoor-fullcolor-160mm-160-mm-1-16/1605560988.html

As far as I can understand, the led matrix will work with your library (32x32, 1/16 scan). What I am not sure is about the led matrix definition (that is, if the text/image will be displayed fine). I know that the P5 defines the pixel size, but I cannot compare it with anything I have seen before.

Based on your experience, whatd o you think about theos modules? Any other advise/suggestion?

Thanks in advance.

maddogs, 2016-05-03:
The Videos below show 64x32 P5 matrix running on Henners Code using an Active Adaptor. All panels purchased from AliExpress.

4 x 64x32 P5s

https://youtu.be/WfIgvow4nlY

6 x 64x32 P5s

https://youtu.be/KsYHHFzNA44

1 x 64x32 P5

https://youtu.be/_PPndQ3sXLE

soyxan, 2016-05-03:
That looks nice!

We are planning to build a big display (well, maybe not very big: 2 rows of 10 32x32 modules). Do you think that it is better to use 64x32 modules (so, 2 rows of 5 64x32 modules)? Is it worth the price?

On the other hand, where did you get the Active Adaptor from?

Thanks!

maddogs, 2016-05-03:
If you hunt around Ali Express, you may find that its cheaper to buy the 64x32 panels. They are internally two 32x32 panels joined together, so one less ribbon cable to break or fault find.

Henner provides the Gerber FAB files to get your own PCB made and a list of components that you need to solder on the front page. However..... I do have a spare blank board, IDC Connectors, Capacitors, Resistors I can send you. You'd need to purchase 4x 74HCT245 SOICS and get somebody to solder it all together. The SOICS are very cheap and can be purchased easily. I did offer this to somebody else on here, but they never got back to me. Let me know.

soyxan, 2016-05-03:
I think it is a good option to buy it from you. Maybe we can arrange it by private.

Going back to the buying guide, I've found this module:
http://www.aliexpress.com/item/320-160mm-3528-HUB75-1-16-scan-indoor-full-color-SMD-rgb-p5-led-module-32x64/32566618174.html

It looks nice, but I have realized that the brightness level is is 2200 cd/sqm. I found other more expensive modules that claims 3000 cd/sqm, but it looks almost the same. Does it really get more brightness?
http://www.aliexpress.com/item/high-resolution-led-matrix-display-module-p5-smd-indoor-fullcolor-160mm-160-mm-1-16/1605560988.html

Thanks!

maddogs, 2016-05-03:
I couldn't tell you how much brighter they would be.

As for the active board, email me your postal address to the address listed under my proflle here

 https://github.com/maddogs 

and I'll send it out. I don't need any money or payment, see it as a community thing.

maddogs, 2016-05-11:
Did the board arrive?

soyxan, 2016-05-11:
I was out of my home till today. I hope that it is waiting form me in my mailbox. I will keep you informed.

soyxan, 2016-05-12:
The board and components arrived perfectly.

Now its time to solder them (not very used to SMD soldering, but I will try).

Thanks!

maddogs, 2016-05-12:
For the Shifters, watch a few Youtube videos and use the slide method they show. It was the first time I did those as well.

maddogs, 2016-05-12:
Another tip. When you solder the 4 Connectors on, make sure get them the right way round! I didn't ;)

See https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter

hzeller, 2016-06-14:
Looks like things got resolved.

Thanks @maddogs to offer your spare boards to other users of the library, that is very kind of you!

@soyxan Did your LED matrix work ?

closed 2017-07-30 by hzeller