hzeller/rpi-rgb-led-matrix #133: HUB75 Adapter

soyxan, 2016-05-05:
Hi,

I have found this kind of HUB75 adapters which integrates several HUB75 connectors and 74HC245, so it *could* be used as an active adapter with level shifting?

http://www.aliexpress.com/item/Hub75-card-Adapter-hub40-hub94-hub75-b-hub40a-hub07-hub17-hub41-hub127-hub100-hub-HUB-08/611324511.html

The interface used is a 50 pin standard used by several control cards manufacturers, and I found a manual where they show this interface pinout (page 9):

http://www.kiasell.com/store/M-10.pdf

What do you think? Could it work just connecting the Raspberry Pi outputs to the proper pins on this 50 pin interface?

Thanks!

hzeller, 2016-05-05:
You need HCT not HC to be able to level shift (the 'T' stands for 'TTL-level'). But you could replace the chips I suppose (they are like < 20 cents each).

hzeller, 2016-05-05:
Regarding the interface: you would need to trace out what the 50 connector inputs are connected to where and where they end up on the outputs.
Closing this for now, let me know if you have any questions when it arrives.

closed 2016-05-05 by hzeller