hzeller/rpi-rgb-led-matrix #134: image viewer

Dan7766, 2016-05-05:
Tried to install the image viewer after the main program and get the below:

NOTE: selecting "libgraphicsmagick++1-dev" instead of the virtual package libgraphicsmagick++-dev"

After this it will not compile the image viewer as i think its looking for files that are not there due to the above Note: downloading the wrong package..

Well done on this by the way, just learning this rite now but having fun....Thanks.

hzeller, 2016-05-05:
This is a normal message of the package manager, so that should be fine.

What is the compilation error you get ?

Dan7766, 2016-05-06:
If its normal perhaps im doing something wrong..
Attached a pic of the files that were not found during libgraphicsmagick install. the rest installed ok. And when trying to compile.
Im sure you will probably laugh at what vie missed or done wrong but hopefully you can see why it wont work for me.

[image]

hzeller, 2016-05-06:
It looks like it didn't find packages on the debian servers. Did you do 
  `sudo apt-get update`
before the `sudo apt-get install` ?

Dan7766, 2016-05-06:
Thank you very much, that worked and the image viewer compiled and works like a charm..
Im guessing i should have known to do that....Looks like i have a lot of reading to do..

Thanks again

hzeller, 2016-05-06:
Don't worry, I have now updated the README to include that line in the description so it is harder to forget.

closed 2016-05-06 by hzeller