hzeller/rpi-rgb-led-matrix #135: No rule to make target 'led-image-viewer'

luchadorman, 2016-05-06:
When I run this,
` sudo apt-get update &&
 sudo apt-get install libgraphicsmagick++-dev libwebp-dev &&
 make led-image-viewer`

It ends with make: ***No rule to make target 'led-image-viewer'. Stop.

I'm realllllllly trying to get this running to put on my graduation cap tomorrow and would love some help!

hzeller, 2016-05-06:
In which directory are you in ?
You compile this at the toplevel directory,

luchadorman, 2016-05-06:
I am in ~/rpi-rgb-led-matrix-master

Is that correct?

hzeller, 2016-05-06:
Yes, that is perfect. If it doesn't work, then you have the wrong repository :)
Make sure to get things here: https://github.com/hzeller/rpi-rgb-led-matrix

luchadorman, 2016-05-06:
Is there anyway for me to check the repository? I was following that documentation during installation.

luchadorman, 2016-05-07:
I am very low level on skills for a project such as this.

hzeller, 2016-05-07:
If you checked out this repository, you're good and it should work. If you got the repository from somewhere else (e.g. Adafruit), it won't work.

luchadorman, 2016-05-07:
I did follow [this adafruit tutorial](https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi/driving-matrices) 

Do you know what I would need to do differently to run this one instead?

hzeller, 2016-05-07:
Do not follow the Adafruit tutorial. They point to their fork of this tutorial, which is ancient and does not have the image viewer for instance.

You are filing issues for this repository, so you should as well check out this library:
https://github.com/hzeller/rpi-rgb-led-matrix.git

Then follow the README. If you have an Adafruit HAT, look in the section "If you have an Adafruit HAT" what to edit in the Makefile to make it work with that hardware.

luchadorman, 2016-05-07:
Ok! I will do that and check the adafruit hat section.

Thank you for your help, I will comment here if I have anymore issues!

luchadorman, 2016-05-07:
I just cleared out my previous files and redid it with your repository and still received the same message. I've attached a picture below.

[image]

luchadorman, 2016-05-07:
I ran `wget https://github.com/hzeller/rpi-rgb-led-matrix/archive/master.zip`

then `unzip master.zip`

then `cd rpi-rgb-led-matrix-master/ && make`

then tried a couple of demos that ran,

and finally ` sudo apt-get update &&
 sudo apt-get install libgraphicsmagick++-dev libwebp-dev &&
 make led-image-viewer`

luchadorman, 2016-05-07:
I've resolved this problem, i cloned the git instead of using wget and everything now works.

closed 2016-05-07 by luchadorman

Skid32, 2016-08-17:
Can you help me wen i run a demo i just get this 
[image]

vic-cortes, 2017-08-24:
Hello, I've been doing the steps that you already mention. I clone the most recent repository from [GitHub](https://github.com/hzeller/rpi-rgb-led-matrix) and followed the steps. However I still have the message `***No rule to make target 'led-image-viewer'. Stop.` 
Is there another way to install the led-image-viewer? 

Thanks in advance

hzeller, 2017-08-24:
Are you in the `utils/` directory ?

vic-cortes, 2017-08-24:
I was in the `rpi-rgb-led-matrix/` directory, but I changed it to `utils/` directory and everything worked fine. Thank a lot.

8bignic8, 2025-11-07:
> unzip master.zip

That worked thanks :)!