hzeller/rpi-rgb-led-matrix #136: Problem with 16x32 DMD ,RPi-3

alperbalim, 2016-05-09:
Hi;
I have problem with using library.You can see that on vide and photos. I have Rpi-3 Model B and 16x32 RGB led matrix
My panel works with both 3.3v and 5v. And I run it with active adapter and without adapter. In the attached video I tried;
`sudo ./led-matrix -D 1 -c1 -r16 runtext16.ppm`

I tried all solutions in the troubleshooting an I could think of. And I read all closed and open Issues. I have still same results. 

What could be the problem?
Video: [https://drive.google.com/file/d/0B6Ea0eHR9iw_ZU1RRFFKTjR2SVYwYTRicERqWFVMME9qWllN/view?usp=sharing](Video link)

[image]
-
[image]

maddogs, 2016-05-09:
What happens if you change the -c 1 to -c 2

alperbalim, 2016-05-09:
Same problem texts and animations can not seem. But I can use full panel now.

currentlysober, 2016-05-09:
First of all you should try to print some text, e.g.
sudo ./text-example -f fonts/8x13B.bdf -y1 -x1 -c1 -r16 -P1 -C0,0,255
and then print e.g HELLO WORLD - in this way we will see, maybe the panel needs some transformer.

alperbalim, 2016-05-09:
Result;
[image]

alperbalim, 2016-05-09:
What kind of transformer panel needs?

currentlysober, 2016-05-09:
hi! did you write "HELLO"? can you write some smaller font, e.g. -f fonts/4x6.bdf ?
so far this is quite different transformation than standard panels.. maybe someone else here has some idea.

alperbalim, 2016-05-09:
Yes I wrote HELLO and tried all fonts. But i couldnt see any text.

alperbalim, 2016-05-09:
I have 4 days to finish that. I am trying 7/24 all ways that I can think of but didn't get any result yet. If anyone can help I will be grateful

KentWalker, 2016-05-09:
It looks like your panel uses a 'snake' type of scan.  I did post a transformer that worked for me somewhere in the comments but I can't find it.  I have attached my modified transformer .cc & .h files...
[snake8x2 trasformer.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/256000/snake8x2.trasformer.zip)

Replace the original files in the /lib and /include directories (but backup the originals first).

see this for an example... 
[test_display.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/256056/test_display.zip)

Recompile (make) the software.

If you run the ledmatrix program will need to add the new transformer to the code.  You need to use rows = 8 and chains = double the number of panels you have in a chain so for a single 32 x 16 panel...
sudo ./led-matrix -D 1 -c2 -r8 runtext16.ppm

alperbalim, 2016-05-10:
I replaced your*.cc and *.h file and tried ./led-matrix and ./text-example. But still I couldnt get any result. You can see in photos;

The command I used;
`./text-example -f fonts/8x13B.bdf -y1 -x1 -c1 -r16 -P1 -C0,0,255`
I sent by the order : "A", "B", "C", "ABC"
[image]
[image]
[image]
[image]

alperbalim, 2016-05-10:
Anybody have any idea?

currentlysober, 2016-05-10:
hi! I have no idea, the way of transforming pixels is strange. Maybe anyone else here..

KentWalker, 2016-05-10:
It looks like the panel is upside down, as x=0,y=0 is in the top left corner.  There will be arrows on the rear of the panel.  You should be plugging into the left socket looking from the rear.

You are pretty close, try changing to -r16.  

If that doesn't help, start with no transformer and -r16 -c2.  Draw a single pixel at 0,0 and note where it is displayed. Keep incrementing x by 4 (leave y=0) and note its x-location until the pixel reaches the other end of the panel.  Don't worry about its y-location at present.  If the pixel only gets half way before disappearing then -c1.  If you get 2 or 4 pixels at once change -r4, -r8 or -r32. Once you have the x and r sorted, do the same again with x but increment y by 1 until it reaches the bottom of the panel before changing x.  You will soon see the mapping pattern and should be able to create a transformer from there.

alperbalim, 2016-05-11:
I tried that with;
`./text-example -f fonts/8x13B.bdf -y1 -x1 -c2 -r8 -P1 -C0,0,255` with a point "."
It starts from top right corner. I tried it other characters.
But display does not change significantly. Also I dont have any idea to create transformer. I checked several times wiring. It seems normal. But still I have doubts. Could it be due to wiring?

Or Does anyone have another idea?

alperbalim, 2016-05-11:
Some character results;
Code: `./text-example -f fonts/4x6.bdf -y0 -x0 -c2 -r8 -P1 -C0,0,255`

Char: "."
[image]

Char: "-"
[image]

Char: "_"
[image]

Char: "O"
[image]

Char: "H"
[image]

Char: "T"
[image]

Char: "/"
[image]

Char: "Z"
[image]

Char: "U"
[image]

Char: "X"
[image]

Char: "L"
[image]

KentWalker, 2016-05-12:
Try this simple test...
[test-pixel.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/260422/test-pixel.zip)

Compile (make) this and run...
sudo ./test-pixel

Play with the number of rows (line 48) until you get just one pixel at a time.  
Leave chain = 2. It won't matter if only one panel attached and stops extra 'x' pixels if the panel is a 'double' type panel.
Change the delay timing in line 29 to suit you.  Recompile each time.

Once you get single 'y' pixels you should be able to create a list/map, eg 
set(0,0) -> display(0,8)   - ie the code 'canvas->SetPixel(0,0... );'  lights a pixel at (0,8) on the panel
set(1,0) -> display(1,8)
.
.
.
set(30,15) -> display(15,15)
set(31,15) -> display(15,15)  

Don't worry if pixels don't appear for every set point or if only half the panel is used at present.  The transformer (which we will create) will fix that later.

alperbalim, 2016-05-12:
`make: *** No rule to make target 'test_pixel.o', needed by 'test_pixel'.  Stop.
`

I got this error

alperbalim, 2016-05-12:
I solved with adding this to makefile;

`test_pixel.o : pixel-test.cc
	$(CXX) -I$(RGB_INCDIR) $(CXXFLAGS) -c -o $@ $<`

alperbalim, 2016-05-12:
I tried all row values but I couldnt get single pixel. Here is video:
8 rows / 2 chains 

[Videolink](https://drive.google.com/file/d/0B6Ea0eHR9iw_OHVCeHJzZ3pQbzQ/view?usp=sharing)

KentWalker, 2016-05-12:
Oops, sorry about the typo in the makefile.  

Looking at the video, it appears the matrix is arranged in the 'snake' configuration - ie it is arranged in blocks of 8x4 pixels and snakes thru half the panel (blocks 1->16) thus:

2 |  4 | 10 | 12
1 |  3 |   9 | 11
6 |  8 | 14 | 16
5 |  7 | 13 | 15

It looks like row = 8  and chain = 2 is correct.

I think the individual block addressing is a bit more complex, so instead of setting a single pixel at (0,0) (which transforms as pixel (0,4) the panel is controlling pixels (0,4), (0,5), (0,6) and (0,7) at once.  Likewise (8,0) transforms as pixels (0,0), (0,1), (0,2) and (0,3).

Try running the test again but change line #23 
'for (int x = 0; x < **64;** x++) {' 
 and I think the other half of the display will light as well (which will prove the chain = 2).

Keep playing with the boundaries in lines 22 and 23, I think your solution can be found.

alperbalim, 2016-05-13:
I am using transfromer that you sent me before. Now the other half of panel is working. But still I can not get single pixel. How can I fix that and create a new transfromer?

KentWalker, 2016-05-13:
What happens if y > 15? If you set (8,16) do pixels  (0,0), (0,1) & (0,3) light or do no pixels light or something else?  Is  the panel actually a HUB75 or is it a HUB8?

alperbalim, 2016-05-14:
I tried `canvas->SetPixel(8,16,170,85,255);` But there is nothing.

I think HUB75 
R1-G1
B1-GND
R2-G2
B2-GND
A-B
C-D
CLK-STB
OE-GND

halwebguy, 2016-05-16:
For what it's worth, I believe I'm having the same issues.  My panel is also 1:4 outdoor, and my H, for example, is split between rows 4-8, and then 13-16.  Should I run these same tests?  

[image]

halwebguy, 2016-05-16:
Also this might help. HH seems to use the alternate rows for the second H (1-4,9-13) but in the same x coordinate space:
[image]

And adding a third H moves to 5-8 and 13-16 in the next x coordinate space:
[image]

This should help with the mapping.  I have no idea what to do from here though :)

Best,
Hal

halwebguy, 2016-05-17:
It didn't  look so bad above , seemed reasonable.  But this is the strangeness I see when I run the test-pixel above.  Link on youtube because the video file is too large:

https://youtu.be/EVyhs9S2_eQ

alperbalim, 2016-05-17:
Still I couldnt solve..

alperbalim, 2016-05-17:
I tried test-pixel with 16x16 panels. I think I am very close to solution. 
` 
rows:8
for (int y = 0; y < 8; y++) {
    for (int x = 0; x < 32; x++) `

But I dont know how to create new transformer 

Here the [video ](https://drive.google.com/file/d/0B6Ea0eHR9iw_enYyYWZCNlp0bU0/view?usp=sharing)

alperbalim, 2016-05-17:
How can I use test_display? I can not compile tehre are a lot of error

alperbalim, 2016-05-17:
Or how can i implement Snake transformer to text-example

KentWalker, 2016-05-18:
Have a look at the txt file here. It contains a template for creating a new transformer.
[newTransformer.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/269410/newTransformer.txt)

Use the Snake8x2Transformer class as an example on how to remap the pixels. Note also the height and width functions are also modified within this transformer class which may be necessary with your transformer class.

I have included a 'fixed' test_display.cc here.  Note I have not compiled to test it.  Change **MyNewTransformer** in line 30 to suit.  I am sure you should be ale to fix and debug any other errors.

alperbalim, 2016-05-18:
I couldnt run test_display. How can I apply new transformer to  text_example ?

alperbalim, 2016-05-18:
I found that. But now I have another problem. How can I mirror chars? And I want to apply that transformer to ./led-matrix command. 
[image]

KentWalker, 2016-05-19:
Wow, that's interesting!  Can you upload your transformer class?  

I'm not sure if this will work but no harm in trying:

Add to bdf-font.cc the following function...

`int Font::DrawMirrorGlyph(Canvas *c, int x_pos, int y_pos, const Color &color,
                    uint32_t unicode_codepoint) const {
  const Glyph *g = FindGlyph(unicode_codepoint);
  if (g == NULL) g = FindGlyph(kUnicodeReplacementCodepoint);
  if (g == NULL) return 0;
  y_pos = y_pos - g->height - g->y_offset;
  for (int y = 0; y < g->height; ++y) {
    const rowbitmap_t row = g->bitmap[y];
    rowbitmap_t x_mask = 0x80000000;
    int x_width = g->width - 1;
    for (int x = 0; x <= x_width; ++x, x_mask >>= 1) {
      if (row & x_mask) {
        c->SetPixel(**x_width - (x_pos + x)**, y_pos + y, color.r, color.g, color.b);
      }
    }
  }
  return g->width;
}
`

The code in bold should mirror the font.

Also add to the grahics.h header file in the public section of the Font class...

`  int DrawMirrorGlyph(Canvas *c, int x, int y, const Color &color,
                uint32_t unicode_codepoint) const;
`

You will also need to add the function DrawMirrorText in this header as well...

`int DrawMirrorText(Canvas *c, const Font &font, int x, int y, const Color &color,
             const char *utf8_text);
`

Finally add the function to the graphics.cc file...

`int DrawMirrorText(Canvas *c, const Font &font,
             int x, int y, const Color &color,
             const char *utf8_text) {
  const int start_x = x;
  while (*utf8_text) {
    const uint32_t cp = utf8_next_codepoint(utf8_text);
    x += font.DrawMirrorGlyph(c, x, y, color, cp);
  }
  return x - start_x;
}
`

Now replace all calls to DrawText() in your code  to DrawMirrorText() and see what happens.

alperbalim, 2016-05-20:
It worked but; I can draw only one char. I am very confused. I want to aplly same transformer to ./led-matrix. It didnt work too. I recorded a video with runtext16. 

here video: [Link](https://drive.google.com/open?id=0B6Ea0eHR9iw_U2xVeDBYaXZnaUE)

my main-demo.cc file: [Link](https://drive.google.com/file/d/0B6Ea0eHR9iw_c2dfb3o1czVnMVU/view?usp=sharing)

my text-example.cc file: [Link](https://drive.google.com/open?id=0B6Ea0eHR9iw_S2dyNkE1VlZDUFU)

my transformer.cc file: [Link](https://drive.google.com/open?id=0B6Ea0eHR9iw_VTlNNVNRUDZKT3c)

alperbalim, 2016-05-20:
I also tried pixel test with transformer funckiton that you sent.
Here the video and pixel-test.cc file;
[Video](https://drive.google.com/open?id=0B6Ea0eHR9iw_enYyYWZCNlp0bU0)
[pixel-test.cc](https://drive.google.com/open?id=0B6Ea0eHR9iw_M09WRWNwS05hblk)

I will be online tonight. If you have a change to look theese I can reply your questions in minutes. @KentWalker 

Thank you. I appreciate for your help

alperbalim, 2016-05-20:
Sorry the video is older. New video is here:

Video : [Link](https://drive.google.com/open?id=0B6Ea0eHR9iw_WUFvbkNvTVMwNzQ)

halwebguy, 2016-05-21:
Thank you, everyone!  For my issue, the Snake transformer fixed everything.
https://youtu.be/8hd5TsyfZLg

alperbalim, 2016-05-21:
Can you upload your codes?. I couldnt fix yet.

halwebguy, 2016-05-21:
@alperbalim I'm sorry, but I don't think my problems are the same as yours.  All I did was add the Snake transformer.

alperbalim, 2016-05-22:
I created right trasnformer for my 16x16 panels. Test-pixel works right. I added a video of test. 
[Test Video
](https://drive.google.com/open?id=0B6Ea0eHR9iw_LTB4eVE0UnV5TnM)My transformer code;

`  int new_x;
  int new_y;
	 if(y<8) {new_y = y%4;}
else if(y>7) {new_y = y%4 + 4;}

	 if(y%8 < 4)
	 {
		 if(x<8) {new_x = 15-x;}
	else if(x>7) {new_x = 39-x;}		 		 			 
	 }
	 

	 if(y%8 > 3)
	 {
		 if(x<8) {new_x = 7-x;}
	else if(x>7) {new_x = 31-x;}		 		 			 
	 }
`

But I have another problem now. When I try text-example or led-matrix, only first panel works and texts and images are mixing. 

`./text-example -f fonts/8x13.bdf -y0 -c2 -r8 -c5  -C0,0,255`
In this photo I wrote "AL" it looks good because it fits to one panel.;
[image]

But here it messes; I wrote "ALP"  and result;

[image]

And video of `./led-matrix -D 1 -c5 -r8 runtext16.ppm`
[Video](https://drive.google.com/open?id=0B6Ea0eHR9iw_NmhJQVY5RUM1U00)

Where I made the mistake?

alperbalim, 2016-05-22:
I added this for transformer to demo-main.cc and text-example.cc

`    LinkedTransformer *transformer = new LinkedTransformer();
  canvas->SetTransformer(transformer);
  transformer->AddTransformer(new Snake8x2Transformer());`

for

KentWalker, 2016-05-23:
Have a look at lines:
**if(y<8) {new_y = y%4;}
else if(y>7) {new_y = y%4 + 4;}**
new_y will always be 0..7   This is OK when you are using only 16 pixels high but you will end up over-writing these pixels if you ever extend your pixel height.

The transform for **y** in Snake will still work for you, it does the same thing but takes into account y > 15:
**new_y = ((y / 8) * 4) + (y % 4);**

You have the same issue with **x**, once x > 15 your maths is turning to custard.

```
if(y%8 < 4) {
  do this 
} else if(y%8 > 3) {
  do that
}

```
can be better written as 
```
if ((y & 4) == 0) {
  do this
} else {
  do that
}

```
which also takes care of y > 15.

So to transform **x** we need to break it into stages we can follow;
1. The transform will repeat in blocks of 8 x 2 (or one panel) so we need a block_offset of 32. [ (x / 16) * 32 ]
2. We need a modulus of 8 when we are subtracting since the smallest number we are subtracting from is 7 so we need an index_modifier. [ 7 - (x % 8) ]
3. When (x/8) is odd we need to add an odd_offset of 16. [ if((x / 8) & 1) new_x += 16 ]  
4. When (y&4)==0 we need to add a y_modifier of 8. [ if(y & 4) == 0) new_x += 8 ]

Therefore our **x** transform becomes:
```
// new_x = block_offset + index_modifier + odd_offset + y_modifier
new_x = ((x / 16) * 32) + (7 - (x % 8));
if((x / 8 ) & 1) new_x+= 16;
if((y & 4) == 0) new_x += 8;

```

shoke2k, 2016-07-20:
Hello, I used your test_pixel to know the mapping of my rgb matrix, but...
To use all panel I used: `rows=32 chain 1`
And I need to change value of "y" at the loop "for" to 32 for use all leds.
`for (int y = 0; y < 32; y++) {
for (int x = 0; x < 32; x++) {`
And this is the result [https://youtu.be/XwYB2tc7cZU](https://youtu.be/XwYB2tc7cZU)
Help me, please :S

rodrigosalinas, 2016-08-21:
Hi @alperbalim . I have also used your test_pixel to determine my mapping. Thank you very much. In fact, here it is:

    if (x < 8) {
        if (y < 4 || (y > 7 && y < 12)) {
            new_x = (x * -1) + 7;
            new_y = y;
        } else {
            new_x = x + 8;
            new_y = y - 4;
        }
    } else if (x > 7 && x < 16) {
        if (y < 4 || (y > 7 && y < 12)) {
            new_x = (x * -1) + 31;
            new_y = y;
        } else {
            new_x = x + 16;
            new_y = y - 4;
        }
    } else if (x > 15 && x < 24) {
        if (y < 4 || (y > 7 && y < 12)) {
            new_x = (x * -1) + 55;
            new_y = y;
        } else {
            new_x = x + 24;
            new_y = y - 4;
        }
    } else if (x > 23 && x < 32) {
        if (y < 4 || (y > 7 && y < 12)) {
            new_x = (x * -1) + 79;
            new_y = y;
        } else {
            new_x = x + 32;
            new_y = (y == 4 || y == 12) ? y : y - 4;
        }
    }

I'm aware it still needs simplification but, as I am not sure it's complete, I'll leave that for later.

I'm using this mapping for my 16x32 P10. When I run your test_pixel, with the transformer updated of course, I achieve to scan all pixels in one panel. My question is: that's all I need? Do I just need to map for one panel? I have implemented my algorithm in the transformer, then I have use it in demo-main and in text-example. All works fine but always just for one panel. I have 2 chained but the second just shows the same thing as the first one. For example, when I run the following command `sudo ./led-matrix -D 1 runtext16.ppm -r 16 -c 2`, I can see the Hello World Leds duplicated in my 2 chained panels.

¿What do I need to do to make the text scroll from one panel to the other? Am I using the led-matrix command in a wrong way? Or my mapping is incomplete? Or is it something else?

I hope I have made myself clear. Once again, thank you very very much.

Dzepuks, 2016-09-02:
I have the same probem. 
I am a beginner. Vai kads var pastāstit kā to salabot. 
From the correspondence I find it difficult to understand the need to do and what not to . 
Please help

Dzepuks, 2016-09-02:
I have the same probem. I am a beginner. Can anyone tell how to fix it. From the correspondence I find it difficult to understand the need to do and what not to . 
Please help!!!

gokmenkeser, 2016-09-30:
The same problem continues, too. That solve the problem? or any other software? thanks..

hzeller, 2016-09-30:
@Dzepuks @gokmenkeser 
You essentially have to figure out where your panel maps which pixel and write a little mapping function for the library to use.

For start, you have to figure out what multiplexing it uses. If it has 32 rows for instance,  you run a demo that exercises all the LEDs and check if `--led-chain=4 --led-rows=32` light up all the LEDs, and then go down with `--led-rows=16` and stop where not all LEDs are lit anymore. Say you arrived at --led-rows=16. Then you know that internally this is  mapped as double long panel with 16 pixel height. So  you can continue with `--led-chain=2 --led-rows=16`.

Next step is to determine where pixels actually end up. For that, you can write a little program that systematically goes through all pixels, or you write some text on the panel and try to determine how pixels end up on the display. That you use as input to form an idea how the transformation works. Typically it is some kind of 'a band of pixels reflecting at the end of the panel' or 'having a band of pixels snaking around somehow'.

Then you can write a transformer, a class that you need to implement that tell the matrix where some input pixel ends up on which output pixel - in the discussions above, you see various examples (for a start, start looking at https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/transformer.cc ).

With this background information in mind, read the whole discussion above. The problem is typically very specific to your panel as there are various mappings out of the ordinary and the library author does not have all the various things (that would be quite expensive). So in the discussion above, you might find some solutions that are not fitting your particular panel, but you'll learn how to write the transformer yourself.

Some overview how to apply your transformer in  your application you find here: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates

hzeller, 2018-01-21:
Please check out the new option `--led-multiplexing` https://github.com/hzeller/rpi-rgb-led-matrix#multiplexing
It does the mapping built into the library (and can also be used from Python.)
(You need to do a fresh checkout and compile, this just went in).

closed 2018-01-21 by hzeller