hzeller/rpi-rgb-led-matrix #137: Led matrix withouth 74HC245 or 74HCT245

soyxan, 2016-05-11:
Hi,

I have found several matrix panels on china that *apparently* does not use the 74HC nor 74HCT ICs.

Instead they use some DP5020-6A2203 ICs which I am not able to get information on google.

I am wondering if they will work with 3.3V or will need level shifters.

I am attaching a picture sent from one of those suppliers where you can see the back of the pannels.

Thanks!

[image]

[image]

hzeller, 2016-05-11:
I don't know either. I guess there is only one way to find out :)

halwebguy, 2016-05-11:
This is a little disconcerting given that I have 6 outdoor panels from
Leeman that just came in the other day, and I can't even read the number on
mine, but it's not 74HC245, 7HCT245, nor the ones you see.  fingers
crossed. I also have another 4 32x16 P3s on order from them. *sigh*

hzeller, 2016-05-11:
always consider using the buffer anyway no matter the input chips on your
LED board, as you want to avoid straining the output buffers of your RPi
driving longer lines.

The choice of not using your own level converter is cutting corners which
you can get away with if the input drivers can't reliably handle 3.3V
levels. Nothing will break if you try it without, it just won't work
reliably; cheap and safe to try out.

So nothing disconcerting; if you want to have a reliable setup, use $2
worth of interfacing logic, and you're good.

KentWalker, 2016-05-11:
Assume all panels accept 5 volt inputs, that seems to be a 'standard' with the majority of LED panel manufacturers and most (if not all) commercial LED display driver boards are 5 volts too.

The 12 DP5020-6A2203 ics and the 4 'CHIP-ONE' ics are probably custom LED driver chips designed to reduce the parts count on the board and propriety so there's little chance of getting info on them at present.

As Henner states, you can't go wrong using the level converter. His PCB design is available as bare boards from OshPark (which ships internationally - I'm in New Zealand), its cheap, easy to assembly and works!  Use 74HCT245D (Philips), MM74HCT245DW (Fairchild) or CD74HCT245M (Texas Instruments)  ics and you will have no problems.

andygrace, 2016-06-06:
Actually all the CHIPONE ICs are documented online.  They make hundreds of millions of them.   Mostly just improved versions of standard 16 channel drivers, but they also do a version of an HCT138.   Here's the specs link.  
http://www.chiponeic.com/index.php/website/products/led
There are datasheets out there - have used before and from a personal point of view they work pretty well.

hzeller, 2016-06-14:
Closing this bug now. In general, the recommendation is to always use a level shifter, not only for the 5V logic aspect but also to better separate the fragile GPIO output buffers from the LED matrix.

closed 2016-06-14 by hzeller