hzeller/rpi-rgb-led-matrix #138: led-image-viewer flashes image on matrix

jmark1m, 2016-05-12:
I am using a 32x64 display (same as 2 32x32 displays side by side I believe).

First of all, all of the demos work great using -c 2 -r 32.  So much is working correctly.

The led-image-viewer however flashes the image up on the matrix and then clears, and then repeats.  This is about once per second with the image staying on the screen at about a 25% duty ( about 250ms on 750 ms off )

I am running the image viewer with -c 2 -r 32 for my 32x64 display.

I am using a 128x64 bit jpg image which looks like successfully scales to 64x32.

I am using a Raspberry Pi 3.

Thanks.

hzeller, 2016-05-28:
So in some other issue we found that there is the sound-system enabled these days in Raspberry Pi installations. That uses the same hardware subsystem as this library, so they will interact badly.

Can you check out issue #150 and see if explicitly disabling sound helps in this case ?

hzeller, 2016-06-14:
Did removing the SND modules help ?
Also, I recommend to have a minimal installation such as the Raspbian Jessie Lite; apparently, some distributions with a GUI have problems some times.

closed 2016-06-14 by hzeller

jmark1m, 2016-06-14:
It appears to be working with the SND modules off.  Thanks.