hzeller/rpi-rgb-led-matrix #141: Missing function: GetPixel(x, y)

BorisFR, 2016-05-18:
Hello.
I'm using your lib and it is fantastic. I am taking pleasure working/playing with it, it simple and it's just working. Your code is clean and understandable and with given samples it is easy to do what I want. Thank you.

One function I'm missing is this one: (in framebuffer.cc)
**Color Framebuffer::GetPixel(int x, int y);** 

I am aware that the original color is "lost": The SetPixel() function uses MapColor() that changes the value provided...

hzeller, 2016-05-18:
Yep; if you want something like GetPixel(), you need to keep your own buffer to have that readily available.

BorisFR, 2016-05-18:
The problem I have is that I'm using Font. So I need to change "int Font::DrawGlyph()" to write to my buffer too. It is not 'clean' ;)

hzeller, 2016-05-18:
This is why Canvas is an abstract interface. You can build an delegating Canvas, that just passes SetPixel() and other functionality to some underlying 'delegatee' (in your case: the RGBMatrix), and also keeps track of the color frame buffer in a way that it is easy to provide GetPixel().

The DrawGlyph takes a Canvas implementation, so it will just work with that.

```
  RGBMatrix matrix(...);
  CanvasImplementationWithGetPixel myCanvas(&matrix);
  // ...
  rgb_matrix::DrawText(&myCanvas, ...);
```

BorisFR, 2016-05-18:
For sure! Thank you!!! (but why I have not thought of that alone?)

closed 2016-05-18 by BorisFR