hzeller/rpi-rgb-led-matrix #142: Full brightness flicker every 5 seconds

yreiser, 2016-05-21:
I currently have the problem that all matrix leds go to full brightness (only for the blink of an eye) every 5 seconds. Is there any help for this? Beside this, there is no flicker and a very stable image.
I'm using the Adafruit hat (pins 4 and 18 linked), an 64x32 Adafruit Panel.

hzeller, 2016-05-22:
So you see a normal bright image, but then a very bright 'glitch' ? I have not heard about that yet.
Does it happen with the led-image-viewer, or with all of the demos ? What Pi are you using ? Is your Pi running regular frequency or overclocked (overclocking is actually bad and brings some internal timings in the SoC out of whack).

Some questions come to mind that _could_ contribute
  - Is there other, very compute intensive stuff running on the Pi so that there could be 'stealing' some CPU resources ? Maybe try the 'isolcpus' trick mentioned in the README.
  - Is there some other library running that might need the PWM hardware (any audio output ?)
  - Is the power supply stable enough and can supply the current needed ?

yreiser, 2016-05-22:
Hi! Thank you for the fast reply.
The glitch is exactly as you wrote... normal bright, then a very short, very bright image (around every 5 seconds).
It's happening in Python (e.g. myMatrix.Fill(20,20,20) with nothing else already shows this problem).
I'm using a regular Pi3, with a fresh Jessie install (nothing else added).
I already tried 'isolcpus=3', but it doesn't make any difference.
No audio output.
Power supply is stable 5V 8A (attached to the 64x32 Matrix).

rhildinger, 2016-05-22:
By any chance do you have the built-in wifi connected to an access point?

hzeller, 2016-05-22:
So you mean it only happens in Python and not with any of the other c++ based programs ?
I suspect that Python might do regular garbage collection which could lead to trouble. Maybe there is some initialization sequence that is best in Python; In that case, @Saij might be a better person than me to help narrow down the problem.

yreiser, 2016-05-23:
Well, I suppose it's only in Python (with rather dark pixels)... the demo's are all quite bright, so it's difficult to see if the problem also appears in C.
Yes, my Pi was connected to a WIFI access point, but when disabling Wifi, there's no difference.

rhildinger, 2016-05-23:
The reason I was asking bout wifi is that I noticed a similar issue when running the led-matrix demos on my RPi3 the other day. Every few seconds, the 64x32 matrix I was using would dim or flicker slightly. The current would dip (say from 4 amps to 3.75 amps) in time with this flicker. Note that I was not using python at the time, but just the native led-matrix demo.

In addition to the flickering I noticed a slight occasional hand in text entry from ssh shell session. In the end the culprit appeared to be the wifi driver (or supplicant software) that was causing the problem. When I switched back to using ethernet only, the problem went away. The software support for the built-in wifi is still very poor at this point.

Saij, 2016-05-24:
@yreiser 
Du you tried it with the native C++ implementation? If not could you please check if the glitch is still happening?
Also could you please send me some example code to look into?
On my Pi2 I didn't saw any brightness glitch at all. Maybe its a specific Pi3 problem?

yreiser, 2016-05-24:
This is a very basic python script, but the every 5s flicker is heavily noticeable (at least with my setup):

from rgbmatrix import RGBMatrix
import time
myMatrix=RGBMatrix(32,2,1) #32 rows, 2 chained displays, 1 parallel chain
myMatrix.Fill(10,10,10)
time.sleep(20)

yreiser, 2016-05-24:
I'm glad I can tell you that my problem is solved !
in fact, I wasn't aware that, after changing options in "/lib/Makefile", I had to do a "make clean" first before the new "make". 

So I had enabled the "Adafruit Hat Options" in the Makefile, but since I always omitted the make clean instruction, I was always driving my Matrix on the Adafruit hat but with the Hat option still disabled. And I wasn't aware that I also had to additionally redo the "make clean" / "make" procedure in the /python folder after changing options in /lib/Makefile.
Anyway, I have a perfectly stable matrix image now with no flickering at all.

Maybe we should add more detailed step by step instructions to the main readme for newbies like me.
If I understand well, this is what everybody has to do in order to get everything up an running:

1) download the repository with "git clone https://hzeller/rpi-rgb-led-matrix"
2) "nano rpi-rgb-led-matrix/lib/Makefile"     then change required options here
3) in folder "rpi-rgb-led-matrix", execute "make"
4) in folder "rpi-rgb-led-matrix/python", execute "sudo python setup.py install"
5) reboot so the python modules are available

Am I right?
And thanxxx to everybody that provided help on my issue!

Saij, 2016-05-25:
In step 4 you don't have todo `sudo python setup.py install`. Instead do `sudo make install`. This is better because the Makefile will do some other things as well maybe.
And you don't need a reboot. After installing the python module it is there without a reboot.

hzeller, 2016-05-28:
So in some other issue we found that there is the sound-system enabled these days in Raspberry Pi installations. That uses the same hardware subsytem as this library, so they will interact badly.

Can you check out issue #150 and see if explicitly disabling sound might be the solution to weird flicker ?

hzeller, 2016-06-14:
Changes of the options in lib/Makefile _should_ actually do the full recompile of all related things, but I suspect the needed recompile in the Python directory was the culprit.

Anyway, glad it is working now.

closed 2016-06-14 by hzeller