hzeller/rpi-rgb-led-matrix #143: 6 panels 32x32 each

arnovi, 2016-05-23:
Hi,

Im using a Rpi3 with Adafruit Hat. I woud like to use 6 panels (3x2) of 32x32 leds.
I've seen the -L option but it only works for 2x2panels setup. Is there a simple way for 3x2panels setup?

Here's the back of my setup:
http://imgur.com/2Kzs8cL

Thanks.

davemaster, 2016-05-23:
Greetings,

Check link bellow, it could help You

[http://www.instructables.com/id/DaveWare-Led-Board/](url)

bbarrilleaux, 2016-06-12:
@arnovi can you point me to "the -L option" you mentioned in your question? I'm trying to figure out how to get a 2x2 grid of matrices set up.

arnovi, 2016-06-13:
Thnkas davemaster but this project is not using same hardware/library.

hzeller, 2016-06-14:
I would suggest to actually connect these in multiple chains @arnovi - that way you have a good refresh rate, and a very simple layout that does not need a transformer.

For that, I suggest to build yourself an adapter
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

You can use three chains with 2 LED panels each (`-P3 -c2` option to the demo programs) to have a tall display, or have two chains with 3 panels each (`-P2 -c3`) to have a wide-screen setup.

If you are interested in a transformer, have a look at the code for the 64x64 conversion https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/transformer.cc#L199 - from there you can deduce how you would to a different layout.

closed 2016-06-14 by hzeller