hzeller/rpi-rgb-led-matrix #144: Compute Module IO board support for 5 chains

rhildinger, 2016-05-24:
For those who might be interested, I've committed the first pass attempt at 5 chain support via the Raspberry Pi Compute Module IO board. The CMIO board brings all 53 GPIO pins out to headers, which allows at least 5 and possible 6 chains to be ran from one Raspberry Pi Compute Module. The repository is located here:

https://github.com/rhildinger/rpi-rgb-led-matrix

The code can be enabled by uncommenting the following line in lib/Makefile:

#DEFINES+=-DCM_5_CHAIN_SUPPORT

I have not yet had a chance to test this code on a CMIO board yet, but i have tested it with a regular RPi3, and so far my changes haven't broken anything!   ;-)

hzeller, 2016-05-24:
Instead of two uint32, it is probably more readable if you encode the bits in a uint64. The GPIO registers are consecutive.

rhildinger, 2016-05-24:
I agree that the code would be simpler and more readable using a singe uint64. In fact I debated for a long time the pros and cons of encoding with a single uint64 vs to uint32's. In the end I decided to go with the two 32's for the following reasons:

1. I was going to send you a pul request once the code was tested.
2. I wanted avoid a performance hit for those not needing 5 chain support.
3. It was easier to keep a good percentage of the code separable via #define in support of #2 above.

That being said, If you think it makes more sense to just go ahead and switch the GPIO section over to uint64 coding for all users so that the code is simpler, I can certainly make that change.

currentlysober, 2016-06-29:
Hi, thank you for the code, I was trying to play with it and I got stuck on one point: I am trying to move 4th row from pins 32-37 into 30-35. (cause I just want to use ping 36 & 36 for uart) Can you please tell me what I am missing? I either have some errors or I have wrong colors and part of the screen blank. 

    ```
unsigned int p3_r1          : 1;  //  30
unsigned int p3_g1          : 1;  //  31
unsigned int p3_b1          : 1;  //  32
unsigned int p3_r2          : 1;  //  33
unsigned int p3_g2          : 1;  //  34
unsigned int p3_b2          : 1;  //  35
unsigned int unused_36_37   : 2;
unsigned int p4_r1          : 1;  //  38
unsigned int p4_g1          : 1;  //  39
unsigned int p4_b1          : 1;  //  40
unsigned int p4_r2          : 1;  //  41
unsigned int p4_g2          : 1;  //  42
unsigned int p4_b2          : 1;  //  43
```

```
for (uint32_t b = 0; b <= 13; ++b) {
    if (outputs1 & (1 << b)) {
      INP_GPIO(b+30);   // for writing, we first need to set as input.
      OUT_GPIO(b+30);
    }
}
```

```
/*static*/ const uint32_t GPIO::kValidBits1
= ((1 <<  0) | (1 <<  1) | (1 <<  2) | (1 <<  3) |
   (1 <<  4) | (1 <<  5) |
   (1 <<  8) | (1 <<  9) | (1 << 10) | (1 << 11) | (1 << 12) | (1 << 13)
);
```

currentlysober, 2016-07-04:
It seems like first pins: R1 and G1 are not active.

BennoG, 2016-09-13:
I have solved this on a more extendable way by using multiple clock lines.
by using multiple clock lines you can daisy-chain all the other lines. I ended up in wiring 2 chains out as in this project and stack the next print on top with only the clock line changed. If I want 2 more chains i can simply put on another print and rewire the clock to 1 of the remaining free outputs.
This solution will cost more time to clock data to the displays but because you can start clocking the data in while the PWM pulse is still going you don't lose any time (except for the short PWM pulses)

P.S. I don't know this library starts clocking in new data while the PWM pulse is still on but I do it without any problem.

Benno

hzeller, 2016-09-13:
Yes, this library is clocking out the next row while the PWM pulse is
ongoing.

BennoG, 2016-09-13:
And what do you think of multiple clock lines as I use over here.
You can stack multiple pcb's and only need to rewire the clock line (I use 3 way soldering pads for this, jumpers take up too much room on the pcb)

hzeller, 2016-09-13:
Multiple clock lines are fine, but you are limited by the speed of the GPIO. So for RPi2 and 3 having two clock lines can help, because otherwise they have to be wasting cycles with the slowdown (I am having some update to this project in which I do something with a dual data rate that allows more than double the number of chains. PCB not ready yet). With three chains, essentially all GPIO pins are already in use (except one), but you could use two chains to stack and have six clock lines.
After two clock lines you have diminishing returns however (speed limit of GPIO).

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828