hzeller/rpi-rgb-led-matrix #145: Switching between canvas and led-image-viewer within python script

delectatronics, 2016-05-24:
I am after some advise if possible.  I can successfully display stable images using led-image-viewer within a python script.  The hardware is a RPI3 with a 3x3 matrix of 64*32 RGB panels.

However I am trying to switch between a canvas object and led-image-viewer within a python script and when doing so the led-image-viewer image has a regular flicker about once a second.  I believe it is because the canvas object is in some way still active and interfering with the led-image-viewer instance.  The code I am executing is below if anyone can offer suggestions on whether my deletion of the canvas is correct or I should do this via another means.

```
canvas = self.matrix
/* do some display processing not listed here */

/* clear canvas and destroy */
canvas.Clear()
del canvas

/* display a logo for 10 seconds */
disp = subprocess.Popen(["sudo", "/home/pi/rgbmatrix/led-image-viewer", "/home/pi/rgbmatrix/BullsLogo1.png", "-p 11", "-r 32", "-c 6", "-P 3"])
print "Display Sponsor proc#" + str(disp.pid)
time.sleep(10)
os.kill(disp.pid, signal.SIGHUP)

/* create new canvas */
canvas = self.matrix
```

yreiser, 2016-05-24:
Instead of using led-image-viewer, I simply display my images with some own code.
Example (using pillow module)

from PIL import Image
img=Image.open("/home/pi/rgbmatrix/BullsLogo1.png")
for x in range(img.size[0]):
    for y in range(img.size[1]):
         pixeltuple=img.getpixel((x,y))

delectatronics, 2016-05-25:
Thank you very much for the reply - I have been able to swap successfully between scoreboard and images successfully without any flicker.
Do you have a code sample for dynamically resizing an image to display (without changing the file itself) and displaying animated GIFs using PIL as this was the original reason I was using led-image-viewer.

If anyone does have a solution to the original issue I would be interested in your thoughts.

Saij, 2016-05-25:
In your code you just got a copy of the matrix. But the matrix as such is not destroyed once and so still running. Maybe that's your problem.

hzeller, 2016-06-14:
Yep, there can only be one canvas running at a time on the system.

closed 2016-06-14 by hzeller