hzeller/rpi-rgb-led-matrix #146: 16x32 rgb matrix not running using raspberry pi 2

abhi-27, 2016-05-26:
Hello hzeller! 
Inspired by your blog i decided to take my annual engineering project as 16*32 rgb matrix controlled by raspberry pi  2. I followwed your link from here(https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi?view=all).i connected everything as told but still nothing works.i took a 5v1a adapter,16*32 rgb matrix with exact same hub75 port.i used this link for programming (https://github.com/hzeller/rpi-rgb-led-matrix) .i followed every step but the board never glows up.this is very important for me. Kindly help me with this.this is very important to me as i have to do it in any case.

maddogs, 2016-05-26:
I'd first get a bigger power supply at minimum a 2 amp supply for a 16x32. i'd aim for a 3amp supply though.

Have a quick read on the front page under the section "A word about power"

hzeller, 2016-05-26:
Please don't use the Adafruit article for the connection diagram - it is outdated and wrong. Read the README in the https://github.com/hzeller/rpi-rgb-led-matrix project.

abhi-27, 2016-05-28:
thank you for the link sir but the board still isn't working properly.i am attaching image of my rgb led matrix.only few of the rows are glowing. kindly suggest me what to do next.

[image]

hzeller, 2016-05-28:
Can you show the back of the board and the connector ?

abhi-27, 2016-05-28:
We had an ic TM74HC245 on it. And its hub75 and back 
looks like
[image]
[image]

hzeller, 2016-05-28:
Ah, if it has a HC245, you need a level shifter to convert the 3.3V logic level to 5V and can't directly connect it to the Pi GPIOs. Have a look at the Troubleshooting section in the README.

abhi-27, 2016-05-29:
I have made gpio slow down 2  in the code and even try to comment it out but results are same.
Do i have to use active adapter for it now or is there any way in which i can use level shifter ic ?

hzeller, 2016-05-29:
Yes, if this is a HC245, then you very likely need to make yourself a logic level shifter board (Make one yourself with the provided board or get an Adafruit HAT which is already mostly pre-assembled).

You might try to have a lower voltage on the LED panel to see if it starts working (say 4V ... it is closer to 3.3V, so the digital threshold might be better suited); the colors will be off, so it is mostly only useful to verify.

abhi-27, 2016-05-29:
Thank you . should i follow this scematic
[image]

This has 4 hc245 ics but those for three panel . for one panel we would require only 2 ics?one for r,g,b inputs and other for clock ,strobe etc.

abhi-27, 2016-06-05:
Hello hzeller! 
I did what u suggested .i made an adapter with level shifter.i got lm245 ics and use two of them for one panel.but this time it didnt even glow up.i followed the scematic very carefully.kindly help

hzeller, 2016-06-14:
If you followed the schematic, it should work. I can't check what is wrong remotely of course, I suggest just make the board for which there are the Gerbers available yourself.

And note, you need HCT245 _NOT_ HC245.

hzeller, 2016-06-14:
Closing for now, let me know if you got it working with the recommended PCB layout.

closed 2016-06-14 by hzeller

abhi-27, 2016-06-16:
Hct245 ic is not available to me so i have used sn74Ls245 ics .their datasheets seems to be same . can i use it . will it work ?

hzeller, 2016-06-16:
Using LS might not work to interface properly with the LED matrix if that is using HC chips as input.

In the following graph you see that the LS chips will properly trigger on the 3.3V logic level of the RPi (input high voltage is 2V), but the output with roughly 3.3V will not be good enough to be higher than the 3.7V needed to trigger HC chips. It might work 'by accident' if the trigger levels are close enough, but it is certainly out of spec.

http://www.interfacebus.com/voltage_threshold.html

So using the LS245 is probably not making much difference. In order to do TTL voltage conversion, you need to use HCT245 or AHCT245.

abhi-27, 2016-06-16:
Thank you sir for the fact .actually I am not able to find hct245 in dip pkg ... Is there any other alternative?

hzeller, 2016-06-16:
Texas instrument has these. The AHCT245 seems to be a bit cheaper. Here a couple of links from some common sellers
https://www.digikey.com/product-detail/en/texas-instruments/SN74AHCT245N/296-1476-5-ND
http://www.mouser.com/search/ProductDetail.aspx?R=0virtualkey0virtualkeySN74HCT245N
http://www.newark.com/texas-instruments/sn74ahct245n/logic-bus-transcvr-octal-20dip/dp/50R5869

abhi-27, 2016-06-16:
Thank you ver much sir

abhi-27, 2016-06-19:
I got the ics hct245. Now board is showing only green colored random text (maybe speed of text is too high)in upper half. i am using only one panel so im using only u1 and u2 ics from schematic. is it correct to use this way? for strobe also i took only one pin in u2(scematic). Can using only two ics be cause of problem?

hzeller, 2016-06-19:
Two ICs should be enough. Make sure to properly connect the A->B line to set the direction; otherwise it not only will not work but also destroy the Raspberry Pi pins (they can't tolerate 5V).

All the unused inputs connect to GND.

Other than that, just follow the traces and make sure everything is properly connected. And given that the longer cabeling can have some capacitance, connect change the slowdown option to 2 in lib/Makefile (DEFINES+=-DRGB_SLOWDOWN_GPIO=2).

(I really suggest to just build one of the boards directly as given in the adapters directory, it saves you much hassle).

abhi-27, 2016-06-22:
Finally !! Its running .Text is flowing and colours are spot on. Thank you hzeller sir a lot for your guidance and help.Although there is some small errors like size of the ppm image is too much for the panel so 1/4  of the top part of image is missing.

hzeller, 2016-06-22:
Congratulations that it is working!
Now you can as well use other programs such as the image viewer or write your own. Have fun!

iammenotyou, 2017-02-28:
On this topic of needing to covert from 3.3v to 5v.   I have a question while I'm waiting for my led matrix to arrive.   :)

On adafruit's tutorial for hooking up the displays to the pi it says to use the pinouts as follows:
GND on display to GND on the Pi (blue or black)
R1 on display to GPIO 17 on the Pi (red)
G1 on display to GPIO 18 on the Pi (green)
etc...

Since I'm not great with soldering and more of a software person than hardware and haven't found any clear instructions for how to connect the converter.
So I have 2 questions, 1. could you use 2 logic converters like http://www.robotshop.com/ca/en/8-channels-logic-converter.html (as these are easier to find locally) than the suggested 74AHCT125 chips.
2. If yes, how would the connections go?
GND on display to B0 on converter, A0 on converter to GND on the Pi (blue or black)
R1 on display to B1 on converter, A1 on converter to GPIO 17 on the Pi (red)
G1 on display to B2 on converter, A2 on converter to  GPIO 18 on the Pi (green)
etc.???

hzeller, 2017-02-28:
Do not use the Adafruit tutorial. It is outdated and wrong after the default pinout changed a long time ago. Always use the original wiring described here: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

The logic converter you mention seems to look good. However, you can as well use HCT245 like in the adapter board https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3 (HCT245 do 8 channels instead of 4 channels like the HCT125)

In general: GND is connected directly.
All data cables are going through the voltage converter. They all go towards the matrix, so input of the voltage converter is on the Raspberry Pi side, output towards the Matrix.

You need 13 voltage converter lines for the first display chain, then 6 for each following chain (because ABCD, CLK, STROBE, OE can be shared between all chains).

iammenotyou, 2017-02-28:
Building a PCB like suggested sounds elegant, but above my skill level at the moment.   Just so I'm clear (and hopefully help any other hardware newbies like me) if I were to stick with a converter I would:

Assuming single chain/display,
GND on Pi to GND on display 1.
Strobe on display to B0 on converter, A0 to pin 7 on pi
clock on display to B1 on converter, A1 to pin 11 on pi, 
remainder continue as per your wiring diagram matching Ax to Bx.  And no need to provide power separately to the chip.  And since there is 13 lines convert, covert half via a converter, and convert the other half via secondary converter.

And I must say amazing work, very informative and a big thank you!

BNNorman, 2017-08-03:
I'd like to add my thanks. Bought the Pi boards from OSH Park,  HCT245 Chips etc from Ebay. Soldered up - works just fine (after fixing some badly soldered joints). Before the Pi boards arrived I cobbled a single HUB75 system using 8 channel level shifters, a breadboard and jumper wires - mostly worked enough to check the LED panels. I now have a 64x64 matrix usng two 32x64 panels from Alieexpress at £16 each . For anyone getting the jitters with SMD soldering the trick is to tin one pad align the component and solder it. It then stays put whilst you do the other legs. I used drag soldering and removed the excess witha solder sucker - I also used flux which helped a lot - you may need a magnifying glass to check your solder has flowed nicely..