hzeller/rpi-rgb-led-matrix #148: RGB display goes blank after few seconds when a program is run as systemd service on boot on a Raspberry pi 2(Raspbian Jessie)

shreyansml07, 2016-05-27:
I've created a simple program to display time and date on a 32x64 RGB led matrix connected to a raspberry pi 2(running Raspbian Jessie) using this library.
```
static void gettime() {
  rawtime = time (NULL) ;
  timeinfo = localtime(&rawtime) ;  
  strftime(timehm,128,"%H:%M",timeinfo);
  strftime(datedm,128,"%d/%m",timeinfo);
  strftime(dayw,128,"%a",timeinfo);
}

int main() {

  rgb_matrix::Font font;
  if (!font.LoadFont("/home/pi/rgb/rpi-rgb-led-matrix/fonts/10x20.bdf")) {
    fprintf(stderr, "Couldn't load font\n");
    return 1;
  }

  GPIO io;
  if (!io.Init())
    return 1;

  RGBMatrix *canvas = new RGBMatrix(&io, 32, 2, 1);
  canvas->SetBrightness(50);
  Color color(0, 255, 255);

  const int x1 = 7;
  int y1 = 6;
  const int x2 = 7;
  int y2 = -2;
  const int x3 = 17;
  int y3 = 14;

  while(true) {
    gettime();
    canvas->Clear();
    rgb_matrix::DrawText(canvas, font, x1, y1 + font.baseline(), color, timehm);
    sleep(4);
    canvas->Clear();
    rgb_matrix::DrawText(canvas, font, x2, y2 + font.baseline(), color, datedm);
    rgb_matrix::DrawText(canvas, font, x3, y3 + font.baseline(), color, dayw);
    sleep(3);
  }
  canvas->Clear();
  delete canvas;

  return 0;
}
```
It works fine when I launch it manually as 
`sudo ./timescript`

I then tried to add it as a systemd service so that it runs automatically when I power on my pi. 
`sudo nano /lib/systemd/system/myscript.service`

myscript.service contains:
```
[Unit]
Description=My Script Service
After=multi-user.target
[Service]
Type=idle
ExecStart=/home/pi/rgb/rpi-rgb-led-matrix/./timescript
[Install]
WantedBy=multi-user.target
```

The permission for unit file has been set as
`sudo chmod 644 /lib/systemd/system/myscript.service`

I then enabled this service using
```
sudo systemctl daemon-reload
sudo systemctl enable myscript.service
```

When the pi boots up, time gets displayed for a couple of seconds and then the led matrix goes blank(nothing is displayed). When I check the status of my service
`sudo systemctl status myscript.service`
it says it is running. 

I'm struggling to understand why the RGB matrix would go blank after a couple of seconds when this is run from systemd.

Saij, 2016-05-27:
Maybe some other service or program has taken away the GPIOs.
Maybe try to run it as last service to start or maybe LSB Init.D Script?

shreyansml07, 2016-05-27:
I noticed something odd. This works fine on boot if a HDMI display is connected. But if I switch the display off and reboot, the RGB matrix goes blank after coming on for a few seconds!

Saij, 2016-05-27:
Then maybe there is a framebuffer that tries to connect to a SPI display.

shreyansml07, 2016-05-27:
I'm sorry, I don't understand that. What do I need to do to solve this(assuming the cause is known)?

Saij, 2016-05-27:
I can't tell you that sorry ^.^
Don't have that much know how on this subject ^.^

hzeller, 2016-05-27:
There is no known reason.
You have to figure out what systemd is doing that is messing with the program.
Does it call something after this program that resets the GPIO ports ? The ways of systemd are mysterious (I usually avoid it for anything on my system because it is too erratic and obscure at this point).

shreyansml07, 2016-05-28:
Alright, so I thought I'll avoid systemd. I reinstalled Raspbian Jessie on the SD card, cloned the library and added the code to run the runtext demo in /etc/rc.local

`sudo /home/pi/rpi-rgb-led-matrix/./led-matrix -D 1 /home/pi/rpi-rgb-led-matrix/runtext.ppm -c2`

The 'H' of 'Hello' barely lights up for a brief moment and then the display goes blank again! Are you able to run any of the examples on boot? If so, could you please let me know how?

hzeller, 2016-05-28:
So in some other issue we found that there is the sound-system enabled these days in Raspberry Pi installations. That uses the same hardware subsytem as this library, so they will interact badly.

Can you check out issue #150 and see if explicitly disabling sound might work ?

shreyansml07, 2016-05-31:
That did the trick! It works fine now. Thanks a lot @hzeller and @Saij for helping me out.

closed 2016-05-31 by shreyansml07