hzeller/rpi-rgb-led-matrix #150: raspberry pi 3

jroelofs, 2016-05-28:
Is there some trick to getting this library to work with the Pi 3? I can make the samples work on Raspbian on a Pi 2, but I don't see anything on the panels when using a Pi 3.

hzeller, 2016-05-28:
Mmh, it worked for me straight away on the Pi3, so not sure what the  trouble is you are running into.

hzeller, 2016-05-28:
Does your Raspberry Pi have audio subsystems enabled by default ? This needs to be off because it uses a similar hardware subsytem.

jroelofs, 2016-05-28:
Excellent suggestion!

I added these modules to the blacklist:
```
$ cat /etc/modprobe.d/raspi-blacklist.conf
blacklist soundcore
blacklist snd
blacklist snd_pcm
blacklist snd_pcsp
blacklist snd_bcm2835
blacklist snd_timer
blacklist pcspkr
$ sudo update-initramfs -u
$ sudo reboot
```
And now it works flawlessly.

closed 2016-05-28 by jroelofs

jroelofs, 2016-05-28:
@hzeller btw, that saved me about $250 on this project I'm building... so if you find yourself in the Denver area sometime, give me a shout and I'll buy you a beer! (I'll definitely share photos of it if you're interested)

hzeller, 2016-05-28:
Wonderful to hear that you got it to work. I think I need to update the documentation if this is now a thing that the Pi comes with the audio stuff enabled.

Photos would be great. If you also have some 'action shot' of your installation, that would be great.

jroelofs, 2016-05-28:
The thing I'm building is a timing & scoring system for Underwater Hockey<sup>[1]</sup>

Assembly of the display cards:
http://i.imgur.com/Qz6rm9N.jpg
http://i.imgur.com/xiiv2wK.jpg

And one of them in its waterproof enclosure:
http://i.imgur.com/lKgVTwi.jpg

Obviously the software is still in the works ;)

1: https://vimeo.com/6919536

halwebguy, 2016-05-29:
Wow @jroelofs , do you have plans for that waterproof enclosure?  I want to build one and have no idea how.  Not the slightest.  

I wrote this, BTW, in Python.  It's cycling though a bunch of scores I'm pulling down.  It's just building animated gifs and then playing them by invoking the app through the system.  I'm working on making it realtime on purely Python, but first I need support for Transformers.

https://youtu.be/IWXTlw2ZO_0

Let me know if you want the code.  I'm not ready to post it publicly yet, not happy with it yet, pretty raw :)

jroelofs, 2016-05-29:
We're trying to turn this into a product (albeit with veerrry low volume) so we're keeping the plans secret for now.... I might be able to sell you one ;)

The basic design is a bunch of glued acrylic. We send that to a machine shop to have the O-ring groove added, bolt holes drilled, as well as holes for ip68 power/antenna bulkheads. The display card itself is laser cut acrylic, with PCBs from OSH Park.

http://navisjon.com/

hzeller, 2016-05-29:
Cool. Very nice.

Make sure when you have this as a product to provide your and the libraries' source-code to the receiver of this product (including your possible changes) to stay within the requirements of the GNU public license. But when you publish it anyway when you're ready, that qualifies.

Interesting that the antenna works under water (I suppose that is the goal ? To have the display under water so that the parties can see the score ?); I thought under water antennas don't really work. Maybe have a serial interface come out and connect  on the outside of the pool. Or maybe even have an infrared receiver and 'broadcast' the display value under water with a strong infrared diode, so you don't need a cable connection.

(I hope the power supply is _not_ submerged within the displays but rather 5V is send down the wire...).

jroelofs, 2016-05-29:
> Make sure when you have this as a product to provide your and the libraries' source-code to the receiver of this product (including your possible changes) to stay within the requirements of the GNU public license. But when you publish it anyway when you're ready, that qualifies.

Definitely. I was kind of hoping your library was LGPL, but given that it's GPLv2, yeah, I know I have to distribute the source.

The antenna isn't going to work underwater, but there will be a 10' long cord from the bulkhead to the surface, and then an antenna attached to that. It turns out that IR doesn't work underwater either :)

And yeah, it's 12v down to the display with a beefy switching supply in the display itself... to keep all of the high voltage away from the pool, and behind GFI circuits.

Definitely a challenging and fun project.

regular, 2016-05-29:
Nice project! I wonder if In-Off-Keying using ultrasound would work for underwater communications in such a turbulent environment.

tacbanon, 2016-11-25:
@halwebguy  Just want to ask if you can share your python code from this video https://youtu.be/IWXTlw2ZO_0?  This will help me start my own project. 
(degoriojoseph@gmail.com)

jroelofs, 2016-11-25:
@tacbanon here's the source for mine, if you find it useful: https://github.com/Navisjon/uwh-display

tacbanon, 2016-11-25:
@jroelofs  Wow!!! I believe I found my new hobby :) this is so cool!