hzeller/rpi-rgb-led-matrix #151: lib hangs in WaitPulseFinished

regular, 2016-05-28:
I had no luck with running the various demos – the matrix stayed black. I noticed that the process would not exit after hitting the return key, so I investigated using gdb.

Thread 1 hangs in join(), Thread 2 in WaitPulseFinished(). I wondered what would happen if I just commented-out the busy-waiting while loop.

```
/* 
   while ((pwm_reg_[PWM_STA] & PWM_STA_EMPT1) == 0) {
       // busy wait until done.
     }
*/
```

To my big surprise and delight everything then worked perfectly! I get ~260 fps for demo 0 and all the others work as well. I am running Arch on RPI2B and the matrix is a 32x64. Isn't this odd? 

I even set `+DEFINES+=-DRGB_SLOWDOWN_GPIO=0` and have no timing issues whatsoever.

hzeller, 2016-05-28:
This is very interesting ... and should'nt happen :/

Do you have anything going on that occupies the audio-system ? Because the audio sub-system also uses the same PWM hardware, so they could get into each other's way.

Try to uninstall anything that has things to do with audio.

depili, 2016-05-28:
The newer raspbian images have the following device tree overlay enabled by default:

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

hzeller, 2016-05-28:
Ok. so that might need to be switched off. If that can be confirmed (for my last Pi, it just worked without), I'd love a suggestion what to add to the documentation.

hzeller, 2016-05-28:
See also issue #150 - that sounds very similar and was fixed by removing the sound subsystem.

regular, 2016-05-29:
Yes, I guess ls the problem. If I do `lsmod`, I get this:

```
# lsmod
Module                  Size  Used by
bcm2835_gpiomem         3718  0
bcm2835_wdt             3621  0
bcm2835_rng             2199  0
rng_core                8227  1 bcm2835_rng
uio_pdrv_genirq         3591  0
uio                     9277  1 uio_pdrv_genirq
sch_fq_codel            7988  2
snd_bcm2835            22483  0
snd_pcm                89187  1 snd_bcm2835
snd_timer              22123  1 snd_pcm
snd                    63597  3 snd_bcm2835,snd_timer,snd_pcm
ip_tables              12494  0
x_tables               16682  1 ip_tables
ipv6                  370596  26
```

I did `modprobe -r snd_bcm2835`, which unloaded *all* of the `snd` modules.

```
# lsmod
Module                  Size  Used by
bcm2835_gpiomem         3718  0
bcm2835_wdt             3621  0
bcm2835_rng             2199  0
rng_core                8227  1 bcm2835_rng
uio_pdrv_genirq         3591  0
uio                     9277  1 uio_pdrv_genirq
sch_fq_codel            7988  2
ip_tables              12494  0
x_tables               16682  1 ip_tables
ipv6                  370596  26
```

led-matrix still hangs. I then blacklisted `snd` by putting this into `/etc/modprobe.d/blacklist.conf`

```
$ cat /etc/modprobe.d/blacklist.conf
install snd /bin/fail
```

and rebboted. `lsmod` confirms there are no sound modules loaded. (same output as above), but led-matrix still hangs.

regular, 2016-05-29:
I then blacklisted nearly all of the other modules, except bcm2835_gpiomem because I assume it is needed by the library? No luck. Then I also blacklisted gpiomem and rebooted.

```
➜  ~ lsmod
Module                  Size  Used by
uio                     9277  0
ip_tables              12494  0
x_tables               16682  1 ip_tables
ipv6                  370596  26zz
```
led-matrix still hangs :( I am out of ideas now.

hzeller, 2016-05-29:
This is strange. Is there anything else running on the Pi that could conceivable mess with the PWM subsystem ? Some other LED library or something ?

What is the Pi you are using ? Pi 1, 2, or 3 ? Which is the particular operating system image you are using here ? I'd like to see if I can reproduce that.

(btw, gpiomem is not needed).

regular, 2016-05-29:
It's [ArchLinux on Raspberry PI 2](https://archlinuxarm.org/platforms/armv7/broadcom/raspberry-pi-2)

regular, 2016-05-29:
Could be the serial console. I tried to stop all services with a name related to getty, agetty and serial.  Alas, I don't really know what I am doing. Could not stop this particular one due to its strange name:

```
# systemctl|grep getty
system-serial\x2dgetty.slice
```
(\x2d is a dash, right?)

hzeller, 2016-05-29:
mmh, so the Pin for the serial port TxD only gets used when parallel=3 (and RxD only ever with rare 1:64 LED panels). Also from my experience it should not affect the PWM.

For the Archi Linux installation: Do you have a link to the particular image (with date) you are using ? They seem to have names such as ArchLinuxARM-rpi-2-latest.tar.gz but I assume they also have filenames with a full timestamp.

Saij, 2016-05-29:
Maybe it is possible to make an image of your direct system for testing (and upload the image somewhere)?

hzeller, 2016-06-14:
If you can't figure out what is broken in your distribution @regular , I recommend using Raspbian Jessie Light which is what I usually do without problems.

regular, 2016-06-14:
@hzeller thanks. Yes, I wasn't able to figure out what's going on. Changing the distro would be an option. Alas, everything works fine when I comment out the while loop, so actually I am fine. @Saij I would have to sweep through it and remove lots of proprietary stuff – it is also pretty large by now. I would be surprised though if the problem would not occur on a freshly installed Arch for RPI2 install, so if you are curious to dig deeper into this, you should be able to. Thanks for your support!

hzeller, 2016-06-14:
If you comment out the while-loop, then it wouldn't wait until the pulse is actually over, so you might not get the right colors.

closed 2017-07-30 by hzeller