hzeller/rpi-rgb-led-matrix #153: Trying to play animated gif

ckng2000, 2016-06-05:
Hi.  i am newbie to raspberry and led rgb matrix but i managed to runtext, other demo and also customize some ppm file with text and graphics and make it run. But when i try out your image viewer there is nothing display on my screen. I am using one panel 32 x 32. Can you take a look and advice what should i do? Thanks

pi@raspberrypi:~ $ cd rpi-fb-matrix/rpi-rgb-led-matrix/
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ sudo ./led-image-viewer myimage2.gif
Read image...
Assembling animation with 8 frames.
Scale ... 32x32 -> 32x32
Preprocess for display.
Display.

hzeller, 2016-06-05:
It looks good and should just work. So if it doesn't, maybe your gif can't be properly decoded ?
Do other images using the led-image-viewer work ?

ckng2000, 2016-06-06:
So far i have tried a few images (animated gif, normal gif, and jpg) all same. Or do you have any working gif that i can try? I use GIMP to scale the image and save as gif maybe that's why is cause the problem?

hzeller, 2016-06-06:
You don't really have to scale the image beforehand, the utility does it
itself. Literally all images I tried have worked so far, so something else
must not be working.

Did you check the "Troubleshooting" section of the Readme? There can be
problems if the sound subsystem is triggered, and maybe the multimedia
parts of the image library do that?

Make sure that snd modules are not loaded by the kernel - The Readme
section suggest how.

ckng2000, 2016-06-06:
I just follow your instruction step by step, did not realize there is a sound subsystem, will give it a try to disable the it. Will let you know the outcome. Thank you so much for the guilines and advice. Apprecate it.

ckng2000, 2016-06-09:
Hi. Sorry to trouble you but i still have no luck making it work. I read through the README.md and there is no mention of sound subsystem/ snd module? I am referring to the 32 x 32 rgb led matrix panel by right there should be no coding related to sound in the demo right?

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Jun  9 07:48:43 2016 from sp3
pi@raspberrypi:~ $ cd rpi-fb-matrix/rpi-rgb-led-matrix/
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ nano README.md
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ ls
adapter  demo.gif      fonts      img      led-image-viewer     led-image-viewer.o  Makefile            myimage2.gif  powerbar.jpg  README.md      runtext.ppm
COPYING  demo-main.cc  hub75.jpg  include  led-image-viewer.cc  lib                 minimal-example.cc  myimage.gif   python        runtext16.ppm  text-example.cc
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ pwd
/home/pi/rpi-fb-matrix/rpi-rgb-led-matrix
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ sudo ./led-image-viewer hub75.jpg
Read image...
Scale ... 154x300 -> 32x32
Preprocess for display.
Display.

hzeller, 2016-06-09:
This is the section:
  https://github.com/hzeller/rpi-rgb-led-matrix#bad-interaction-with-sound
It was only submitted a couple of days ago, so maybe you don't have it in your git copy ? Do a `git pull` to get the latest.

Sound is nothing this library does, but the Raspberry Pi might provide sound due to some other reasons; and the sound-modules might hog that piece of hardware infrastructure the rgb-matrix library is using.

hzeller, 2016-06-14:
Did it work ?
Another thing what people have reported is when they use a GUI based distribution. I recommend to just use a Raspbian Jessie Light.

Closing for now unless there is new development.

closed 2016-06-14 by hzeller

ckng2000, 2016-06-14:
Hi. When i did a lsmod in fact something does show up on the snd module. thanks for the detail module coding but as a beginner i don't know how to create the kernel module blacklist file. I did some reading on that part it ask my to install Virtualbox, Vagrant and kennel-o-matic from adafruit. I tried without success so i guess i have to learn more for now till i understand how to create the blacklist file. I really wanna thank you for the detailed instruction. Apart from the image viewer i did learn many new thing from you. Cheers

hzeller, 2016-06-14:
The Troubleshooting section contains two copy-paste commands that should help with creating the blacklist file.

ckng2000, 2016-06-15:
Managed to remove snd from lsmod but still cant get the image viewer working.  I cut and paste my command over. On the top part when i try to display runtext.ppm its working fine but when i change directory to image viewer the command seems to be working but nothing show up.

pi@raspberrypi:~ $ cd rpi-rgb-led-matrix-master
pi@raspberrypi:~/rpi-rgb-led-matrix-master $ sudo ./led-matrix -r 32 -c 1 -d -t                                                     10 -D 4 runtext.ppm
pi@raspberrypi:~/rpi-rgb-led-matrix-master $ sudo ./led-matrix -r 32 -c 1 -d -t                                                     10 -D 1 runtext.ppm
pi@raspberrypi:~/rpi-rgb-led-matrix-master $ cd
pi@raspberrypi:~ $ cd rpi-fb-matrix/rpi-rgb-led-matrix/
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ ls
adapter       hub75.jpg            lib                 python
COPYING       img                  Makefile            README.md
demo.gif      include              minimal-example.cc  runtext16.ppm
demo-main.cc  led-image-viewer     myimage2.gif        runtext.ppm
etc           led-image-viewer.cc  myimage.gif         text-example.cc
fonts         led-image-viewer.o   powerbar.jpg
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ sudo ./led-image-viewer/img/powerbar.jpg
sudo: ./led-image-viewer/img/powerbar.jpg: command not found
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ ls
adapter       etc        include              lib                 myimage.gif   runtext16.ppm
COPYING       fonts      led-image-viewer     Makefile            powerbar.jpg  runtext.ppm
demo.gif      hub75.jpg  led-image-viewer.cc  minimal-example.cc  python        text-example.cc
demo-main.cc  img        led-image-viewer.o   myimage2.gif        README.md
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ sudo ./led-image-viewer powerbar.jpg
Read image...
Scale ... 2048x1231 -> 32x32
Preprocess for display.
Display.
^[[B^[[A^CCaught signal. Exiting.
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ ^C
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ ls
adapter       etc        include              lib                 myimage.gif   runtext16.ppm
COPYING       fonts      led-image-viewer     Makefile            powerbar.jpg  runtext.ppm
demo.gif      hub75.jpg  led-image-viewer.cc  minimal-example.cc  python        text-example.cc
demo-main.cc  img        led-image-viewer.o   myimage2.gif        README.md
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ sudo ./led-image-viewer hub75.jpg
Read image...
Scale ... 154x300 -> 32x32
Preprocess for display.
Display.
^CCaught signal. Exiting.
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ sudo ./led-image-viewer
Expected image filename.
usage: ./led-image-viewer [options] <image>
Options:
        -r <rows>     : Panel rows. '16' for 16x32 (1:8 multiplexing),
                        '32' for 32x32 (1:16), '8' for 1:4 multiplexing; Default: 32
        -P <parallel> : For Plus-models or RPi2: parallel chains. 1..3. Default: 1
        -c <chained>  : Daisy-chained boards. Default: 1.
        -L            : Large 64x64 display made from four 32x32 in a chain
        -d            : Run as daemon.
        -b <brightnes>: Sets brightness percent. Default: 100.
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ sudo ./led-image-viewer -r 32 -c 1 demo.gif
Read image...
Scale ... 32x32 -> 32x32
Preprocess for display.
Display.
^CCaught signal. Exiting.
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $ lsmod
Module                  Size  Used by
bnep                   10340  2
hci_uart               17943  1
btbcm                   5929  1 hci_uart
bluetooth             326105  22 bnep,btbcm,hci_uart
joydev                  9024  0
brcmfmac              186599  0
brcmutil                5661  1 brcmfmac
evdev                  11396  4
cfg80211              427855  1 brcmfmac
hid_logitech            6925  0
rfkill                 16037  4 cfg80211,bluetooth
bcm2835_gpiomem         3040  0
bcm2835_wdt             3225  0
uio_pdrv_genirq         3164  0
uio                     8000  1 uio_pdrv_genirq
i2c_dev                 5859  0
fuse                   83461  3
ipv6                  347530  46
pi@raspberrypi:~/rpi-fb-matrix/rpi-rgb-led-matrix $

dumes20, 2018-01-13:
any resolve on this? @ckng2000