hzeller/rpi-rgb-led-matrix #154: Canvas Transformer on multiple chains

philoepp, 2016-06-09:
Hi,
I understand how the canvas transformer is working (regarding to the sample for the 64x64 screen). But how does the canvas transformer works on mulitple chains? Are the chains internally arranged to a display of the size chain-count*module-height X module-count*module-width?

And if so, is it possible to support a big display with a odd number of modules, where the module count is not a even mulitple of the chain count?

I've set up a 128x128 screen that works lika charm with one chain, but I'm expanding to a 160x160 screen mit 25 32x32 modules, where i would like to use more than one chain, as it would allow a higher refreshrate.

Thanks a lot!
Best Regards
Philoepp

hzeller, 2016-06-14:
If you have multiple chains, then the height of your display is number of parallel chains x height of one display. So say you have three parallel chains with 32x32 displays, each chain 5 panels long, then you have an underlying display of 160x96 pixel.

In you case with 25 modules, you probably would use a chain-length of 9 and parallel 3. You are just not adding the same amount of displays in every chain; so it might look like this .. with `x` being the missing panels
```
x□□□□□□□□ ---|
x□□□□□□□□ ---| Pi
□□□□□□□□□ ---|
```
 It is ok to not have the same number of displays in a chain, but your mapping function now needs to deal with mapping the 160x160 to the 288x96 display, that has two 'holes'.

philoepp, 2016-06-14:
Hi,
thanks for your answer. 
In the meantime I've read your description of the multiple chains and the matrix transformer again. The solution you mentioned here, was the only possible result, if I had thought about it more, before I posted this issue.

I will try this the next weekend when I have some free time.

Best regards
Philipp

hzeller, 2016-06-14:
Ok cool. Closing this issue for now then.

closed 2016-06-14 by hzeller