hzeller/rpi-rgb-led-matrix #155: Adding a Transformer to Python Code

FarhadGSRX, 2016-06-12:
Good day team,

Like many, I have been drawn to this library because of its great utility with RGBMatrices: it's awesome.

Just recently I got another 64x32 matrix, totalling up what I have to a 64x64 matrix.

I see that there is talk of a class that does a transformation allowing two daisy chained 64x32 boards to behave like one large 64x64 square.

Although I have been able to see and observe that functionality while using the demo, I have not been able to get that to work while running any of the Python samples.

How does one invoke or use the transformer in their python code?

Please forgive me if my question is missing something very obvious, I have spent all of 5-6 hours looking through all of your code trying to figure out what that is.

Thanks in advance!
FG

bbarrilleaux, 2016-06-12:
@FarhadGSRX @hzeller Similar(ish) question -- I'm hoping to drive a grid of 4 32x32 LED matrices (as a 64x64 square) using Bibliopixel in Python. Can you confirm that this isn't currently possible? I've been staring at the "LargeSquare64x64Transformer" trying to figure out if there's some way I can access it in python. Thanks for any info.....

FarhadGSRX, 2016-06-12:
@bbarrilleaux Quick question: How does one use BiblioPixel and this rpi-rgb-led-matrix library at the same time? Do we call bibliopixel functions from inside functions written in the usual samplebase.py format? A piece of sample code would be greatly appreciated if that's possible. (I haven't taken too much time looking at BiblioPixel's code myself yet; so if you think this is trivial, just tell me so. Lol. Thanks!)

bbarrilleaux, 2016-06-13:
@FarhadGSRX In the adafruit fork of this library, there's a bibliopixel driver (https://github.com/adafruit/rpi-rgb-led-matrix/blob/master/ada-matrix.py) which uses this library to control an led matrix using bibliopixel. An example python script using the driver looks like this:

`
from ada_matrix import DriverAdaMatrix
from bibliopixel import *
driver = DriverAdaMatrix(rows=32, chain=1)
led = LEDMatrix(driver, 32, 32, serpentine=False) #MUST use serpentine=False because rgbmatrix handles the data that way
 
from BiblioPixelAnimations.matrix import GameOfLife
anim = GameOfLife.GameOfLife(led)

try: 
    anim.run(fps=15)
except KeyboardInterrupt:
    led.all_off()
    led.update()
`

Using this driver, I can write animations using Bibliopixel (such as the GameOfLife one used here as an example), and run them on the LED matrix via the driver. Note that there's a typo in the driver: the dash in the filename isn't valid python when you try to import it, so it needs to be an underscore in the filename. (https://github.com/adafruit/rpi-rgb-led-matrix/pull/6)

... but as to how to extend this into a grid of matrices... That's where I'm lost.

FarhadGSRX, 2016-06-13:
@bbarrilleaux Thanks so much for the explanation! Here's hoping we get some explanation on the 64x64 transformer soon too. =]

bbarrilleaux, 2016-06-15:
@FarhadGSRX FYI with a little help I got this going, using Bibliopixel to remap the pixels into the correct physical arrangement of my panels. Check it out: http://forum.maniacallabs.com/showthread.php?tid=100 

Same thing might work for you, if you're interested in using bibliopixel to write your animations etc.

FarhadGSRX, 2016-06-15:
@bbarrilleaux Agh, amazing. I just wish I were back at home to test this out. Thanks for your help!

closed 2017-07-30 by hzeller