hzeller/rpi-rgb-led-matrix #156: led matrix not working using SSH session

bagumondigi, 2016-06-13:
Hello, when I use the LCD screen via HDMI everything work fine but when I using putty or remote desktop (xrdp) the program not working anymore, nothing happen on led matrix and no error message. Has anyone encountered a similar problem like this ?

bagumondigi, 2016-06-14:
If I using putty or remote desktop when HDMI plug in, the program will work on ssh session. I think when LCD screen was connected, raspberry will load the original session (GUI interface), and it will 'active' all function, system resource.

hzeller, 2016-06-14:
I have never used the Pi with HDMI, always with SSH, but I have never seen this problem.
However, someone recently reported something similar (or was that you ?)

I recommend Raspbian Jessie Lite.

bagumondigi, 2016-06-14:
it seems that is not me, I just open this topic a few hours ago
when we finish the OS installation, we still have to set up static IP using HDMI right ?

hzeller, 2016-06-14:
I usually connect the first time to the Pi by having it get an IP via DHCP, then log in and do the necessary configurations. Alternatively, I use just the serial console.

I have never connected a monitor to it, but I suspect that even without the GUI installation, you can get a login-screen there, so setting up the static IP by modifying the /etc/network files should work.

hzeller, 2016-06-14:
And the RGB Matrix display code on the Raspberry Pi _does_ work with an SSH session without fault (in fact I do that all the time). I just suspect that something else is broken with the GUI system.

I have now updated the README to include the suggestion to use a minimal configuration.

hzeller, 2016-06-14:
Closing for now, let me know if using the minimal Raspbian fixes it.

closed 2016-06-14 by hzeller

bagumondigi, 2016-06-14:
Raspbian Jessie Lite fixed it . This is what i'm done
1.  After write successful, I use monitor to get Raspberry's IP  address (when I scan with Advanced IP Scanner it doesn't give me the IP, i think because i connect it direct to laptop instead router)
[image]
2. I connect to wifi using: 
`sudo nano /etc/wpa_supplicant/wpa_supplicant.conf `
`network={
    ssid="your-network-ssid-name"
    psk="your-network-password"
}`
After this step i can scan Raspberry address using Advanced IP Scanner
3. install git: `sudo apt-get install git`
git init: `git init`
`git submodule add https://github.com/hzeller/rpi-rgb-led-matrix.git matrix`

@hzeller I don't know how to "get an IP via DHCP". Did you mean modifying the /etc/network files ? I use window and it can not read to SD card.

hzeller, 2016-06-14:
By default, the Raspberry Pi gets its IP address on the ethernet interface via DHCP (i.e. it automatically configures it (in your example above, it is 192.168.1.4)). The process to give out an IP address is done by the DHCP service, which is typically running on your router.

I usually ask my router what the devices are which it has given out an IP address recently, so I know which to connect to (the standard username/password of the debian distribution works over ssh then).

bagumondigi, 2016-06-15:
Hello @hzeller , how do you program with raspberry pi via SSH ? Using simple editor like nano and compile code in Raspberry or edit it in PC and then upload to the Pi and compile in it or using cross compiler and just upload executable file to the Pi ? Or another ways ? thanks for reading

hzeller, 2016-06-15:
I usually use emacs in tramp mode. Make sure to set up your ssh so that you can go in without password (put your public ssh key into .ssh/authorized_keys).

Then you can edit things like this:
emacs /pi@192.168.1.100:/home/pi/rpi-rgb-led-matrix/Makefile

Opening new files, saving, even M-x compile will work.