hzeller/rpi-rgb-led-matrix #157: Pi3 Adafruit HAT issue

tehmooo, 2016-06-13:
So i've bought https://makersify.com/collections/leds/ ... -5mm-pitch and https://www.pi-supply.com/product/adafr ... oC-hLw_wcB and it's all hooked up and installed as per the documentation (following yours, not the one from HAT page)  - but when I run the demo 0, I'm only getting reds and greens in the rotating square... and it's a bit jerky. I've soldered the jump wire pin 4-18 and changed the bits in the makefile, disabled audio... still not working

Any thoughts? It's driving me nuts!

hzeller, 2016-06-14:
If you are not getting all the colors, check if your cabeling for the blue is correct (sometimes these IDC connectors loose a connection because it is essentially a press-fit).

It could as well be that there is another process on your Pi that is hogging th GPIO pins, so that the blues are not set ? Make sure that there is nothing else (I usually use a Raspbian Jessie Lite image).

Regarding jerkiness, make sure that you don't have any sound modules activated, see Troubleshooting-section in the README.

tehmooo, 2016-06-14:
So i've now reinstalled with jessie lite, turned off sound modules - http://varlociraptor.com/images/File_000.jpeg this is what i'm seeing on the first demo - just red green yellow. When i run demo 1 with runtext.ppm I see the hello world, then for a split second i see a blue exclamation mark and the smiley face is red...

I've also switched to a Pi2 instead of the 3 to see if that helps, which it hasn't done.

hzeller, 2016-06-14:
Ok, so it sounds like somewhere in the cabeling or the board itself the blue line is interrupted or at least janky because you sometimes see flashes of blue. Do you use a logic level converter to 5V ? Check if the soldering is good. If you don't use a logic level converter then confirm that your board actually can deal with 3.3V logic level by examining the chips on it.

tehmooo, 2016-06-14:
Thanks for replying - regarding the blue, i'm not sure that's the case... because if i run the python script 'pulsing-colours.py' then it cycles through blue seemingly fine, but it doesn't ever seem to do white - just puts red back up. Is this all an issue of using this adafruit hat? Could the panel be faulty?

hzeller, 2016-06-15:
mmh, some LED panels have a non-standard pinout, maybe the blue is on some other pin ?
Do you have a picture of the PCB around the input connector on the LED board - it typically has a silk-screen that describes the functions of each pin.

tehmooo, 2016-06-15:
Thanks for the reply - there doesn't seem to be anything written on it http://varlociraptor.com/images/temp/File_000.jpeg http://varlociraptor.com/images/temp/File_001.jpeg 

after running lsmod i get http://varlociraptor.com/images/temp/File_002.jpeg as the list - is this correct?

Also, i'm running a gui version of raspbian jessie lite... is that an issue?

lastly, if i follow the instructions on the HAT fork they mention to pass -w to the demo string stating that you can use 1 or 2 as the amount to help the panel work, which doesn't work for me - but if i put something like 100 in there (sudo ./led-matrix -r 32 -c 1 -t 60 -D 1 -l -w100 runtext.ppm ) I get a much larger range of colours, but obviously it's really glitchy.

Does that make any sense to you? I don't really understand what that's doing

jamesdouglasappleprint, 2016-06-15:
In addition, I've just tried switching the cable out for a different one and it still has the same issue - so I don't think it's the Pi or the Cable, which leaves the HAT the Panel or a code issue - I dont think it would be the panel since it will display the colours under certain circumstances..

tehmooo, 2016-06-15:
In addition, I've just tried switching the cable out for a different one and it still has the same issue - so I don't think it's the Pi or the Cable, which leaves the HAT the Panel or a code issue - I dont think it would be the panel since it will display the colours under certain circumstances..

hzeller, 2016-06-15:
The `-w` is what `-DRGB_SLOWDOWN_GPIO=1` is in `lib/Makefile`, but I guess you already played with that as you read the Troubleshooting section. You can set that to 1 or 2 (or manually add more to it in lib/gpio.h).

But if it takes more, then this indicates that your RGB matrix is bad - it should just work with that. It is not a code issue as it runs for thousands of installations, so your next step would be to measure with an oscilloscope if you get the right outputs on the blue pin to determine if you have a bad panel or a bad HAT. The interesting thing is, that it is both blue channels in the upper and lower half of the display, which indicates that it is more likely a panel issue. Also, you should confirm with the manufacturer of the panel what the pins are - it is interesting that it is not printed on the back.

Also make sure to have a solid independent connection between GND of the matrix with GND of the Pi.

Anyway, at this point it is a hardware issue that you have to investigate, nothing really I can do remotely. Let me know if you find something interesting.

tehmooo, 2016-06-15:
Thanks for the reply :) - Cool, this gives me a better idea of where to go next - I figured it could be the panel, since no one else seems to have this problem. I'll see about returning the panel this evening and try a different one, see if that helps. I figured it probably wouldn't be the code since I think i've got an isolated issue.

Thanks again for your help!

jamesdouglasappleprint, 2016-06-22:
Thought i'd let you know - fixed it. Powersupply was only producing 2volts and the 5mm pitch panels need a steady 5v to work properly. Thanks for your help getting to the bottom of it!

closed 2016-08-23 by hzeller