hzeller/rpi-rgb-led-matrix #158: Audio with the API?

clgyl, 2016-06-14:
Hello!
First, thanx to a great API!
I have made som great stuff with it and on the current project Im trying is to add sounds to my project.
I have python code which displays different suff on a 32x16 matrix - works perfect.
I tried to load mp3/wav files which plays when the matrix is led but it doesnt seem to work. I have checked the forum and I saw some answers that audio must be disabled.

So my question: Is it not possible to play mp3/wav files and light the matrix on the same time (using headphone as output)?

BR
Clay

Saij, 2016-06-14:
Maybe you could try it with an USB soundcard dongle. 
I don't know if such device also needs the build-in PWM.

hzeller, 2016-06-15:
Sound does _not_ work in parallel with this library because both the same hardware subystem to generate small pulses.

The suggestion by @Saij should work if you can make it only talk to the USB device and not use the bcm snd module. The USB sound adapters are cheap enough and they are likely even higher quality than the built-in sound of the Pi anyway.

### Not recommended other solution below:

You _could_ disable the hardware pulse generation, but you'll get a lot of flicker and brightness gliches with that, depending on the load of the computer, so definitely not recommended, but here it is:

In
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.cc#L407

Replace
```
  if (HardwarePinPulser::CanHandle(gpio_mask)) {
```

with
```
  if (false) {
```

This is not tested, so I don't know if it would work.

In this case you definitely want the isolcpus configuration as mentioned in the README. But anyway, the brightness glitches will not be fun, so you will end up with the USB sound adapter anyway.

clgyl, 2016-06-16:
Thanks for your answers!
I will give it a try with a USB soundcard dongle

BR
Clay

closed 2016-06-16 by clgyl

clgyl, 2016-07-03:
I just want to say that I got the LED matrix to work with sound :). I bought a USB sound card and plugged it in and it worked. So if anyone else want to use play sounds with the LED matrix I can recommend the "Logilink UA0053".
I just plugged it in and edit the 'alsa-base.conf':
`sudo nano /etc/modprobe.d/alsa-base.conf`
edit the line. Use "0" instead of "-2"
i.e.:
`options snd-usb-audio index=-2`
to
`options snd-usb-audio index=0`

reboot the PI and and I played sound with the python line:
os.system("aplay soundfile.wav")

BR
Clay

hzeller, 2016-07-03:
Wonderful, thanks for the feedback!