hzeller/rpi-rgb-led-matrix #162: Not sure but not working with new noobs raspbian install

petreuss, 2016-06-23:
Hello,

There seems to be a problem running the led-matrix with the new raspbian install through noobs.  If I install an older version everything works as it should, but the newest version doesn't push anything to the led board.

Is there any information that I can find to help diagnose this?

hzeller, 2016-06-23:
Did you try the things in the Troubleshooting section ? It seems recent Raspbian installations have enabled the sound-system in a way that it interferes with the hardware components the library needs. The troubleshooting section describes how to disable them.

petreuss, 2016-06-23:
Thank you for your reply.

I'll look into that.

I finally got it working by installing an earlier version of raspbian, but
will certainly have students ask why it's not working with a current
version.

Thank you for what you do.  Your project is an amazing learning experience!

closed 2016-08-23 by hzeller