hzeller/rpi-rgb-led-matrix #163: display on P3 32x64

lucas-pouchot, 2016-06-28:
Hi, can we use your soft to display something on led panel P3 32*64 ?

running : `./led-matrix -D 1 -r 32 -c 2  runtext.ppm`
I got : 
[image]

and with : 
`(while :; do date +%T ; sleep 0.2 ; done) | sudo ./text-example -f fonts/8x13B.bdf -y8 -c2 -C0,0,255`

I got : 
[image]

Lines 8 to 15 are displayed on the lines 0 to 7 and lines 24 to 31 are displayed on the lines 16 to 23.

I'm using a Raspberry Pi 3 and the new wiring schema.

Thanks.

currentlysober, 2016-06-28:
I think you need to write transformer. Or try at first with different row=X values..?

bagumondigi, 2016-06-28:
I used P5 32*64 and it worked fine ! Try this
`sudo ./led-matrix -r 32 -c 2 -D 1 runtext.ppm`
instead
`./led-matrix -D 1 -r 32 -c 2 runtext.ppm`

hzeller, 2016-06-29:
This looks like your `D` address line does not make it through. Can you check that connection ?

lucas-pouchot, 2016-06-29:
Oh, thanks, I used my wiring for the 16x32 panel and I forgot the `D`
Shame on me :D

closed 2016-06-29 by lucas-pouchot