hzeller/rpi-rgb-led-matrix #166: [Praise] Thanks for the PCB design and writing the library

depili, 2016-06-29:
Just wanted to thank you for your good work. Just finished soldering the active adapter today and everything is working fine :)

I noticed that the active adapter section doesn't currently have pictures of a as-built adapter of the current revision with rpi power in. If you want I can submit high res photos of my adapter to you.

[image]
[image]

Seeing how exited other people here at the Helsinki Hacklab are we might be building even more adapters quite soon :)

depili, 2016-07-20:
Here are some high resolution photos of the built adapter.
[image]
[image]

And I hereby place them in the public domain, so you are free to use them on the documentation if you so wish.

Helbrax, 2016-07-26:
Agreed.  This is a great project.  One of the easiest libraries to use and extremely well documented actually.  I didn't use the board, but I designed a smaller one myself to run just one chain.

https://imgur.com/8j5dEKX

hzeller, 2016-07-27:
Thanks for the pictures @depili ! I'll consider using them next time I update the README for the board.

masto, 2016-08-03:
I just put mine together a couple of days ago as well - and it works, thanks so much for designing and sharing it! I agree, these pictures would be good as a gut check to make sure I wasn't putting anything on backwards. I assume it's either/or on the capacitors? This is mine (before I cleaned it).
[image]

depili, 2016-09-27:
Another project, a clock for AV production purposes, uses one 32x32 4mm pitch module and a 60 ws2812b led ring:
[image]

hzeller, 2016-09-27:
Very nice!