hzeller/rpi-rgb-led-matrix #167: Old Raspberry Pi 2 install no longer working

Helbrax, 2016-07-01:
I just dusted off my setup of this from a few months ago.  When I put it away, it was working fine.  Now I am getting similar issues to others that have the snd modules active.  This is an older install of raspbian from December, and has been touched in quite some time(since February or March).

I have tried adding the snd modules to the blacklist, but upon reboot, they are still getting loaded, so I can't seem to disable them to see if that is the issue.

I've double and triple checked the wiring.  Unless there is a bad wire or bad shifter, I'm not sure what else it could be.

Helbrax, 2016-07-01:
Well I managed to disable the snd modules, and I am still having the same issue.  I will double check everything again.

hzeller, 2016-07-02:
For debugging, I just submitted https://github.com/hzeller/rpi-rgb-led-matrix/commit/0ce27d5c7d96e625673af5ca6dac51ccceee4e5e

Can you do a fresh pull and see if the problem goes away if you uncomment the `#DEFINES+=-DDISABLE_HARDWARE_PULSES` line in `lib/Makefile` and recompile ?

It might us help figure out if it is the PWM hardware occupied by sound or some other reason.

Helbrax, 2016-07-02:
I literally just finished getting it to work.  I had to do a fake install of the snd module instead of just blacklisting it.  After that, it was still doing it.  I removed the alsa-base files via apt-get, as well as mpg123.  After that I did a reboot and it worked after that.   Not sure what was going on or if in all that process I "jiggled" something that was loose while looking it over.  Still very strange.

Helbrax, 2016-07-03:
I'm going to mark this as resolved.  If I have any more issues going forward, I will make the suggestions listed above and play around with it some more.

closed 2016-07-03 by Helbrax