hzeller/rpi-rgb-led-matrix #168: Rpi3 Shaking,flickering, jitter

jdbuendia, 2016-07-02:


jdbuendia, 2016-07-02:
Hi, I'm getting a jittery and a upper shadow that not had in the rpi2.

https://www.youtube.com/watch?v=CJf2wXflyRw
https://www.youtube.com/watch?v=7nezryJceOM

The sound is turned off, and the system has no workload.
in rpi 2 it does not occur. Which may be due?

hzeller, 2016-07-02:
The RPi3 is a little faster than the RPi-2, and some displays have trouble dealing with the signal speed. Can you try to set the `DEFINES+=-DRGB_SLOWDOWN_GPIO=1` in `lib/Makefile` to 2 from 1 to see what happens ?

jdbuendia, 2016-07-02:
Hi hzeller, that was, I changed the line to -DRGB_SLOWDOWN_GPIO = 2, and it does not flicker.
Congratulations, this library is a great job.

https://www.youtube.com/watch?v=mGfwIM7u2pw

closed 2016-07-02 by jdbuendia

bagumondigi, 2016-07-20:
hi, @jdbuendia how can you show terminal to LED display in your video ? can you guide me ? Thanks you :)

jdbuendia, 2016-07-20:
hello, take a look this https://github.com/adafruit/rpi-fb-matrix

chrisoutwright, 2017-04-18:
I changed my Pi 3 to boot to command line instead of Desktop and the flicker is much lower, also I enabled internet sharing on eth0 for wlan0 hotspot before, I disconnected eth0 and also flicker is lower. Just moving the mouse produced noticeable flicker so better not to have periphery on.