hzeller/rpi-rgb-led-matrix #169: CPU usage using python library

Helbrax, 2016-07-03:
I'm putting together an application and have implemented the Image Scroller from the demo application into mine.  It works fine, but one thing I've noticed is that the CPU usage is double or sometimes more.  The only major thing I can see different is that the Image Scroller uses a Transformer which it doesn't look like I have access to in python.  I would expect some overhead from using python vs c, but it seems a but more than I would imagine.

Thanks

```python
from rgbmatrix import RGBMatrix
from PIL import Image
import time
import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)

GPIO.setup(21, GPIO.IN, pull_up_down=GPIO.PUD_UP)

rows = 32
chains = 2
parallel = 1

myMatrix = RGBMatrix(rows, chains, parallel)
canvas = myMatrix.CreateFrameCanvas()
myImage = Image.open("YOMxOEb.ppm")
pix = myImage.load()
pos = 0
canvasWidth = canvas.width
canvasHeight = canvas.height

while True:
    if(GPIO.input(21) == False):
        canvas.Clear()
        myMatrix.SwapOnVSync(canvas)
        #add stuff to do here later
        time.sleep(5)
    for x in xrange(0, canvasWidth):
        for y in xrange(0, canvasHeight):
            (r, g, b) = pix[(pos + x) % myImage.width, y]
            canvas.SetPixel(x , y, r, g, b)
    canvas = myMatrix.SwapOnVSync(canvas)
    pos  += 1
    if pos < 0:
        pos = myImage.width
    time.sleep(0.03)```

hzeller, 2016-07-03:
Yeah, the performance in Python is not necessarily stellar, see also #159 
For now, I would suggest to use the C++ API, which is just as nice.

closed 2016-08-23 by hzeller

hzeller, 2016-09-05:
BTW, there is a `SetImage()` now (since last weekend or so) in the Python API. It is not yet optimized, but it might be a little faster than setting pixels manually.
See python/samples/image-scroller.py

hzeller, 2017-01-08:
Thanks to @chmullig, the SetImage() implementation is now fast as well: https://github.com/hzeller/rpi-rgb-led-matrix/commit/6b6d273ec1d3f9883b41f31b3ea51d0969678bbe