hzeller/rpi-rgb-led-matrix #170: RPi 3 + 32x32 Panel, Doesn't work...

joham97, 2016-07-07:
Hey guys,
This is my setup:
- Raspberry Pi 3 (Raspbian Lite)
- RGB LED Panel - 32x32 (http://skpang.co.uk/catalog/rgb-led-panel-32x32-p-1329.html)
- Panel Adapter (http://skpang.co.uk/catalog/32x32-rgb-led-panel-adapter-for-raspberry-pi-p-1344.html)
- Power Supply 5V 2A

I hooked the panel up and see some flickering like this:
http://i.imgur.com/XU1hcIA.jpg

Now I've connected the panel to my Pi and the flickering stops.
But I see nothing, while running "sudo ./led-matrix -t 10 -D 1 runtext.ppm".

Would you kindly note that I'm a newbie in this hole thing. Therefore I think, the solution could be very easy.
But from my point of view, the panel is well powered. What could be my problem?

hzeller, 2016-07-07:
Given that the SK Pang people made this in 2014, it is likely that they use the old pinout. For that, you need to uncomment
#DEFINES+=-DRGB_CLASSIC_PINOUT
in `lib/Makefile` and recompile.

joham97, 2016-07-07:
Sounds great.
I have the same problem, but now the program killed itself after 10 seconds now.
Previously, i had to Crlt + C to kill it.

joham97, 2016-07-09:
Now I use jumper cables to wire it up. Now it's definitely recieves data. I can see some kind of scrolling now, but it doesn't show the right image.
I tried DEFINES+=-DRGB_SLOWDOWN_GPIO=2.
But it doesn't solve the problem.

joham97, 2016-07-10:
Now I got this:
https://www.youtube.com/watch?v=WblwZX8szkk
I see some kind of scrolling.
Any suggestions?

joham97, 2016-07-13:
I'm still helpless.
Without powering the panel, I only got red pixels on. With 5V 2A all LED lights up but only every fourth displays something scrolling.
Tried every opened and closed issue, which could solve my problem, without any success.
I really need your help, I'm about to giving up soon, but I won't.

hzeller, 2016-07-13:
These are the most important things:
 * Did you wire it up with the information in the README ?
 * Did you add level shifters ? Most panels need level shifters and won't work with direct wiring.

joham97, 2016-07-14:
I exactly followed the instructions in the README, multiple times. But the level shifter thing could fix it. I ordered the Adafruit HAT. Could you recommend this level shifter?

hzeller, 2016-07-14:
I would recommend using the level shifter provided in this project.
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3

The Adafruit should work (but remember to change the library compile flags to use the Adafruit pins as they choose a different pinout. Also, you might want to do the hardware modification needed on the Adafruit HAT improve PWM timings. The README explains all necessary steps).

joham97, 2016-07-20:
Thank you. Now I can definitely see the result.
But now the panel looks like this. (c = any color, w = white color)

wwccwwccwwccwwccwwccwwcc
wwccwwccwwccwwccwwccwwcc
wwccwwccwwccwwccwwccwwcc
wwccwwccwwccwwccwwccwwcc
wwccwwccwwccwwccwwccwwcc

There are white rows on the panel, but "behind" them, I can see the right result.

joham97, 2016-07-25:
Any suggestions?

hzeller, 2016-07-25:
do you have a picture of this? e .g. how does it look like when displaying demo 0 ? What are your settings you give to the program ? Do you have a logic level converter?

joham97, 2016-07-25:
Clip of demo 0:
https://www.youtube.com/watch?v=E_yTBEC02Ug

sudo ./led-matrix -t 100 -r 32 -D 0

The Adafruit HAT. 
I did the Makefile changes (DADAFRUIT_RGBMATRIX_HAT and tried the DADAFRUIT_RGBMATRIX_HAT_PWM thing).

hzeller, 2016-07-25:
In that video: is the data cable come in from one of (left or right) or from one of (top/bottom) ?

If it is (left/right), then this looks pretty strange as it essentially means that the shift-register messes up things.

If it is (top/bottom), it sounds like either
  * The multiplexer on the board is probably pretty slow, I would try something like changing the GPIO-slowdown time and possibly the PWM nanos (lib/Makefile).
  * This could be a 1:4 multiplexing board. Try with `-r8 -c4` to see if that results in at least properly colored pixels though a little messed up in location. You need to write a transformer that maps the snake (I should probably add such a transformer to the default transformers).

joham97, 2016-07-26:
The cable is coming in from the right. What is your recommendation to fix this?
Should I try an other panel, e. g. the Adafruit panel?

hzeller, 2016-07-26:
Did you try the `-r8 -c4` paramters and did it change anything ?

joham97, 2016-07-26:
Of cause, I got nearly the same result, But now row 5 to 15 and 20 to 32 are off.

hzeller, 2016-07-26:
Ok, so you are on to something, this is good news.
Write a little program that goes through all the pixels to understand the pixel mapping inside that panel and write a transformer (include/transformer.h).

joham97, 2016-07-26:
Ok, thank you! This gives me great hopes.

joham97, 2016-07-26:
I changed your minimal-example code. I write a program to fade from nothing to green, like this:

for(int i = 0; i<255;i+=1){
   canvas->Fill(0,i,0);
   usleep(20000);
}

I got the first 16 rows in white. the last 16 rows with the same white rows every third and fourth.
Additionally, the 8 pixels, which doesn't displays white changed from Red to Yellow, like the basic color is red and they mixed together into yellow.
https://www.youtube.com/watch?v=sBwhBpneeU4

hzeller, 2016-08-23:
Sounds like you got it working ?

joham97, 2016-08-23:
No, I got no response from my last post...

joham97, 2016-08-23:
I started this project to program a 32x32 led panel. I'm not very into soldering and all this stuff.
The best way for me, would be a all-in-one solution.

hzeller, 2016-08-29:
if you have strips missing like that it could be that you have 1:8 multiplexing for which you have to write a transformer until this is more permanent part of this library.

Only you have your board, so you need to systematically find out where pixels are mapped. See if this is a 1/8 or 1/16 by trying `--led-rows=16 --led-chain=2` (for 1:8) vs. `--led-rows=32 --led-chain=1` (for 1:16) and let something run over it. Write a little program, that goes through pixel by pixel so that you understand what mapping your panel has. You already started with the program, you just need to refine it to understand your mapping.

jeffmjack, 2016-10-09:
I had a similar problem with RPi scrambling LED outputs.
I found that when I hooked up the pins according to the older RPi diagrams (the 26-pin layout) it worked though.

I suppose this would be a hardware-based approximation of what @hzeller suggested with compiling with #DEFINES+=-DRGB_CLASSIC_PINOUT uncommented.

Anyway, maybe try jumping the wires from your cable to the following RPi pins?

| LED pin       | Rpi physical pin | 
|-------|-------|----------| 
| G1     | R1    | 13 | 23 | 
| GND  | B1    | 6   | 26 | 
| G2     | R2    | 21 | 24 | 
| GND  | B2    |      | 19 | 
| B       | A       | 16 | 15 | 
| D       | C      | 22  | 18 | 
| STR   | CLK  | 7   | 11 | 
| GND  | OE    |      | 12 | 

Maybe this implies a problem with the DRGB_CLASSIC_PINOUT? I don't know C so can't really dive any further into the bugfixing.

Hopefully this is somehow helpful...

joham97, 2017-01-05:
It's finally done. The Matrix works as intended and I am super happy about it. 
I bought a RGB LED Panel by adafruit and their HAT to avoid problems discussed above.
I guess I ruined the old HAT with a soldering gun, so I bought a new HAT, a pencil style iron and try it again. And it finally works.
Thank you so much for helping me!

closed 2017-01-05 by joham97