hzeller/rpi-rgb-led-matrix #171: RPi 3 + level shifter (74HCT245N) + 16x32 LED panel does not work

gnltkak, 2016-07-09:
I'm trying to [this tutorial](https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/overview) but does not work.

At first I tried 74AHCT125N for level shifter but it didn't work. So I'm using 74HCT245N now according to README of source code. It seems that there is no problem in 74HCT245N and LED panel. I tried all methods in trouble shooting section, but it did not work also. My LED panel does not light at all but sometimes shine 'randomly'.

Should I try older(1 or 2) RPi ? Or does anyone have a good solution?

hzeller, 2016-07-11:
Raspberry Pi 1 or 2 or 3 all work.

Make sure to connect `CE` and the direction switch `DIR` to the right signal (`CE` needs to be connected to GND, `DIR` to 5V).

Look at the schematic in the `adapter/active-3 directory` for inspiration.

(however, if you have _not_ connected `DIR` to 5V before, you might've destroyed the Raspberry Pi GPIO pins as they have been exposed to 5V input which they can't deal with).

gnltkak, 2016-07-11:
Thank you for your answer.

At first I connected `DIR` to 0 and all level shifter's Bx pins(=input pins when `DIR` is 0) to input pins of LED panel.
But regardless of `DIR`, is it possible that any input voltage goes in GPIO pins under this condition? 
(Because all level shifter's Bx pins were connected to panel's input pins)

gnltkak, 2016-07-11:
I've confirmed all GPIO pins have no problem. I already checked `adapter/active-3` directory and made test circuit like that. When I probe each GPIO pins by multimeter(I have no oscilloscope) with running `led-matrix -D 0`, almost every pin is 0 except OE, CLK, and more about 2 pins (I can't remember exactly).

hzeller, 2016-07-11:
So you probed the pins and they are all 0V, how does that lead you do the conclusion that the GPIO pins do _not_ have a problem ?

gnltkak, 2016-07-11:
I tested each GPIO pin individually with simple python code like `GPIO.output(pin_number, 1)` and serial resistor-LED circuit. GPIO pins are almost 0V when I run `led-matrix -D 0`.

gnltkak, 2016-07-12:
When i run `led-matrix -D 0`, result is like below:

<left side>
2: 3.3V (OE)
3: 3.3V (CLK)
4: 0V (LAT)
17: 0V (R1)
27: 0.3V 
22: 0V (B1)
**10: 1.65V** 
9: 1.65V (C)
11: 0.1V 
**5: 3.3V** 
**6: 3.3V**
**13: 3.3V**
19: 0V
16: 0V

<right side>
14: 0.3V 
**15: 3.3V**
18: 0V (G1)
23: 0V (R2)
24: 0V (G2)
25: 0V (B2)
8: 1.65V (B)
7: 1.65V (A)
12: 0V
16: 0V
20: 0V
21: 0V

It seems that pin connection to panel is wrong. Is it possible tutorial is not up-to-date?

hzeller, 2016-07-12:
The README is up-to-date.

(You hopefully didn't follow the Adafruit tutorial which they have based on some ancient version of the code. If you use that, then you have to enable CLASSIC pinout in the makefile).

hzeller, 2016-07-12:
The Adafruit tutorial actually mentions the classic pinout here:
https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/testing
so with that, it _should_ work.

gnltkak, 2016-07-13:
Adafruit tutorial was completely out-of-date. Pin connection was not correct.
So I connected pin with README and finally it works!
But there exists some noisy flickering. How can I remove that?

https://youtu.be/IPb0hJktYyw

hzeller, 2016-07-13:
Your PI is probably too fast for the Matrix. Try setting the define `RGB_SLOWDOWN_GPIO` to two in `lib/Makefile`.

gnltkak, 2016-07-13:
That is result of `DEFINES+=-DRGB_SLOWDOWN_GPIO=2`.
When `DEFINES+=-DRGB_SLOWDOWN_GPIO=1`, flickering becomes more severe.

hzeller, 2016-07-13:
Ah, then you might need to slow down that even further (by manually adding another slowdown copy in `include/gpio.h`).

But first, try something else, because it typically is not necessary.

First, I'd check that the power is good (that your panels has a solid 5V at the input which means the power supply should be steady and you have connected it with short enough cables, and that you have a low ESR largish capacitor directly at the power input; see 'a word about power' section in the README). Sometimes it helps to power he LED panel with slightly more voltage like 5.5V.
I have seen this kind of flickering in particular when the power is not stable.

Also minimizing and having a roughly equal length of all the data cables is a good idea.

gnltkak, 2016-07-13:
I forgot connecting panel's GND and level-shifter's GND. Please forgive my basic mistake :(.
Thank you for your help and kind explanation!

closed 2016-07-13 by hzeller