hzeller/rpi-rgb-led-matrix #172: 32x64 display does not show green on first 16 rows

zzxydg, 2016-07-09:
Hello,

Firstly thank-you for the great library you have created and the PCB files for the adapters, I found them very easy to use and get setup.  Everything almost seemed to work first time, except for a problem with green!

I am using the original RaspberryPi model B, with raspbian lite (May-2016) and the passive adapter to link the pi to a single 32x64 RGB matrix I got from ebay.  The setup is all powered by a 400W ATX power supply so there is plenty of juice!

The problem is that I cannot get the first 16 rows of the display (1-16) to light green, they light red and blue just fine but not green.  The second 16 rows of the display (17-32) work fine and light red, green and blue really well.  The chips next to the HUB75 connector are marked SM245TS and I've build the library with GPIO slowdown set to 2 but still no green and all the demos run really smooth with no glitches.

Is there anything I can do in the software library to help debug and potentially solve the problem or could the panel simply be defective?

Thank-you again for a great library and PCB files any any help or pointers you can give to resolve this problem.

David.

hzeller, 2016-07-09:
Well, sounds like the green line is not making it through for whatever reason. I don't know if SM245TS is actually 3.3V compatible (see troubleshooting section for experiments, for instance lowering the voltage on the display to 4V to see if the threshold changes; but you need a bench supply for that).

On the PCB not plugged into anything, I'd measure with an ohm-meter if there is not an accidental short to ground on the green line and that the connection from the G2 is actually arriving at the corresponding header to verify that the signal is not lost somewhere.

Maybe your RPi has a damaged GPIO pin ? For that, measure with an oscilloscope if something happens on the green line. If you see it even at the input of the matrix, then the matrix could be defective, yep.

In general, I'd advise to always use an active adapter - even if your boards can deal with 3.3V logic level, it is nicer to the Raspberry Pi to not have to drive long IDC cables directly from the CPU.

zzxydg, 2016-07-10:
Hello,
Thank-you for your help and advice and the prompt response.  Here is what I have found

There are no faults in the wiring, all check out okay
All the signal lines apart from G1 pulse logic high and low (I can test this on the display)
The Raspberry Pi is not outputting any data to G1
If I short G1 to G2 (on the display) then the top half of the display shows green also, so the display is not defective ... good news as I did not want to have to pay the expense of sending back to china!
So the question now is why is G1 (pin 13) no outputting any signal?

Reading on the internet I notice that the Raspberry Pi foundation modified the original model B soon after release and swapped around the GPIO assignments for Pin13 from GPIO21 to GPIO27 and if I check the revision of my model B in python I get the following:-

_Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
>>> GPIO.RPI_REVISION
1
>>> quit()_

So I have the original original model B (revision one) prior to the GPIO swap as I was one of the first customers to receive a Pi from the foundation.

So my question (before I go out and buy a new Pi) is does your library identify the model revision of the Pi and if it is a revision one output G1 to GPIO21 and if it is a revision two output G1 to GPIO27?

Thank-you again for your help and advice is is very much appreciated.

David.

zzxydg, 2016-07-10:
Sorry for second posting on this.  Further testing of my pi has confirmed that pin 13 is okay and working fine.  The code below produces a nice square wave on pin 13

#!/usr/bin/env python
import RPi.GPIO as GPIO
import time
GPIO.setmode(GPIO.BOARD)
GPIO.setup(13, GPIO.OUT)
while True:
	GPIO.output(13, True)
	time.sleep(0.1)	
	GPIO.output(13, False)
	time.sleep(0.1)

With regard to the library, I have been reviewing the source code and can see in gpio.cc a reference to raspberry pi 1 revision 1 on line#86 but it is not clear to me how the library uses this either automatically or forced via a define in the makefile?

Thank-you again for your help and advice getting working with this library, it is very much appreciated.

David.

zzxydg, 2016-07-10:
Hello,

I have solved the problem.  I used your advice to test the hardware and observed that nothing was coming out of G1 so digging into the library I saw that pin13 was being driven via GPIO27 (which works for rev2 but not rev1) so by making a small change to the library to push G1 to GPIO21 rather than GPIO27 the board started showing green on all 32 rows.  The change seems to work very well on my RaspberryPi but probably is not very portable to other Pi versions.

I am new to GitHub and C++ so have not uploaded my changes back into the library as they will certainly break things for other people, but I have attached the modified file so that the changes made could be merged back into a future revision if you feel this would be a good thing. 

[framebuffer-internal.h.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/356178/framebuffer-internal.h.zip)

I only changed two lines in framebuffer-internal.h ln203 and ln212.

Thank-you for your help debugging and for creating a great library.

David.

milluza1, 2016-10-17:
I'm having the same issue on a 16x32 panel now. The first 8 rows are a purplish color, and the lower 8 are green(as they all should be set). I am using the python runtext example as a template. The framebuffer-internal.h looks very different to me (I've loaded this on my Pi in the last month or so). What should I change to fix this issue?

hzeller, 2016-10-18:
The hardware mapping is now defined in the [lib/hardware-mapping.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/da05f01bdba3cfdf613479821eed068b892e6d88/lib/hardware-mapping.c#L45) file.

If you are using the 'regular' mapping, you need to change around line 45 from
   .p0_g1         = GPIO_BIT(27),
to a
   .p0_g1         = GPIO_BIT(21),

(I never had an origina Pi1, so I never could really test that.)

hzeller, 2016-10-18:
Ok, better solution just submitted. 
No, just give the following command line option `--led-gpio-mapping=regular-pi1` and it should work.

closed 2016-10-18 by hzeller

milluza1, 2016-10-18:
Thanks for the fix! It's now working for me if I'm running the C++ examples. However, it seems it's not applying the change when I try to use the python examples. Is there anything additional I need to do for that to work?

hzeller, 2016-10-19:
For Python, you need to compile the libray by setting the hardware description, as you can't yet choose it from the Python lib:
```
    make HARDWARE_DESC=regular-pi1
````

milluza1, 2016-10-20:
Hm, still not working for me. Sorry, I'm somewhat of a noob in Linux, but I tried running that command to compile in /lib and it looks like it ran properly, but alas it hasn't seemed to correct the issue. Is this the wrong directory to compile?

hzeller, 2016-10-20:
Yes, you need to recompile in the Python directory and reinstall that.

milluza1, 2016-10-20:
Hmm, still a no go. I have tried going into /python/ and I first did "make HARDWARE_DESC=regular-pi1" then I followed the install guide and did "make build-python" and then "sudo make install-python." Still doesn't work. Is that the correct procedure to reinstall? Sorry to keep bugging; I just feel so close!

hzeller, 2016-10-20:
Did you make clean before ? You should be able to do everything from the toplevel directory

```
make HARDWARE_DESC=regular-pi1 clean build-python
sudo make HARDWARE_DESC=regular-pi1 install-python
```

milluza1, 2016-10-21:
That was the ticket. It's working great now. Thank you so much!

bphillippi, 2021-02-27:
I have the same problem with no green showing on top 16 rows of a 32x64 matrix display on Raspberry PI 4 and have followed all of the above instructions. Any suggestions?

hzeller, 2021-02-27:
If you're having a Pi4, your problem is different (the Pi1 just didn't have the wire). How is your Pi connected, are you useing the adapter boards provided in this project or some other adapter @bphillippi ? 
Mostl likely the green wire is not connected properly. Make sure that GPIO 27 is properly connected from the Pi to the digital level converter to the Matrix.

commit 24ecf1d8 (2016-10-18): o Provide a "regular-pi1" hardware mapping that should work with
  both, Raspberry Pi1 Revision 1 and 2.
  Fixes #172