hzeller/rpi-rgb-led-matrix #174: Clock example duplicate and flashing

bagumondigi, 2016-07-13:
Hello, i have setup a 128 x 64 pixel screen using 4 modules 32 x 64 in 2 parallel 0 and 1.
other examples work fine including led-image-viewer but when I tried with example clock it occurs duplicate and flashing.
https://youtu.be/DMIRaeO1pL8
this is what i type: 
`(while :; do date +%T ; sleep 0.2 ; done) | sudo ./text-example -f fonts/8x13B.bdf -x 33 -y 24 -r 32 -c 4 -P 2 -C255,0,255`
has anyone ever encountered similar problems yet?
Thanks for reading !

depili, 2016-07-13:
On Wed, 13 Jul 2016, bagumondigi wrote:

> Hello, i have setup a 128 x 64 pixel screen using 4 modules 32 x 64 in 2 parallel 0 and 1.
> other examples work fine including led-image-viewer but when I tried with example clock it occurs duplicate [...]

You need to use -c 8 as each 32x64 module is internally two 32x32 modules.

-- 
This message may contain traces of Truth.

bagumondigi, 2016-07-14:
@depili 
My setup using 2 chain, 2 panel each chain not 4 panel in serial. So it's should be -c 4 (4x32=128, my screen width).
I've tried your way and nothing show.

hzeller, 2016-07-14:
Well, your example works perfectly well.
Please read the comment to the `text-example`, it shows multiple lines if there are multiple lines fitting. Since your display is so large, new lines are printed at the end, until it reaches that, then jumps back to the top.

closed 2016-07-15 by bagumondigi